robotology/mesh-support #1313: [iCubErzelli01] Left hand middle proximal not working properly

ddetommaso, 2022-02-28:
### Device name 🤖

iCubErzelli01

### Request/Failure description

From yarpmotorgui the joint 13 of the left arm does not change position on demand.

### Detailed context

The initial calibration goes well and from yarpmotorgui the motor of joint 13 looks fine, not in hardware fault
[log_icub-head_yarprobotinterface_2511.txt](https://github.com/robotology/icub-tech-support/files/8154501/log_icub-head_yarprobotinterface_2511.txt)
.

gsisinna, 2022-03-03:
The connector of the motor 1B4M1 (middle proximal) was broken and we fixed it. Now the joint works well and we tested it with @ddetommaso.

[image]

cc @Uboldi80

Uboldi80, 2022-03-04:
Following the restoration of the motor flat cable through a cable joint, I realized that unfortunately the flat was interrupted in several points.
@emilianob80  replaced the engine flat along its entire length and this morning I reassembled and tested the forearm.
the proximal middle is now functioning regularly
I re-tested all the joints of the hands, bull arms and head.