robotology/mesh-support #1316: Repair iCubGenova02 - Left arm

sgiraz, 2022-03-01:
### Device name 🤖

iCubGenova02

### Request/Failure description

Left arm no longer powered

### Detailed context

As I was restoring the robot to its state before the tests, I found that all the boards powered by the mc4plus of the left arm had stopped working (no longer powered).

emilianob80, 2022-03-02:
@Uboldi80 @gsisinna Left shoulder EMS4 electronic board replaced as required

Uboldi80, 2022-03-07:
Following an EMS4 firmware update, 3 of the 8 boards failed.
Most likely I imagine that there is a problem on FirmwareUpdater, which during the update of several boards at the same time could have created problems with the bootloader of the same.
@sgiraz  what do you think?
Due to this problem, @gsisinna  and I had to extract the 3 EMS4 (EB1-EB2-EB3) from the back of the robot and reprogram them on the bench with the risk of creating damage to the setup.
I strongly urge you to verify why such a thing happened.

cc. @marcoaccame @Fabrizio69

closed 2022-03-07 by Uboldi80

pattacini, 2022-03-07:
>Following an EMS4 firmware update, 3 of the 8 boards failed.
Most likely I imagine that there is a problem on FirmwareUpdater, which during the update of several boards at the same time could have created problems with the bootloader of the same.

Do we have any hypothesis to inquire @marcoaccame ?

davidetome, 2022-03-07:
Honestly, in the last years, I didn't experience any bug-related issue of this kind; It happened the same to me sometimes but the  2 main issues were : 

- A power supply issue during the programming
- A mistake in the manual procedure of programming the updaterforBootloader/application 

Surely @marcoaccame could have any thoughts..

cc @pattacini @Uboldi80 @gsisinna

marcoaccame, 2022-03-07:
> Following an EMS4 firmware update, 3 of the 8 boards failed.

I haven't ever seen such an extensive failure. It is difficult to say what has happened.
In any case, note the following:
- in case the update of an application on an ETH bord fails for any reason (in the past I surely tested: loss of link, of UDP packets, power off in the middle of an update, programming an ems w/ FW of an mc4plus) then you can recover by getting the ETH board in maintenance. You do that by switching off and on and then by sending a discovery straight away (in the first 5 seconds).
- reprogram the application, maybe one by one so that you can see what happens.

Uboldi80, 2022-03-07:
Of course, the best thing is to take them one by one probably always, avoiding cumulative updates.
Unfortunately, however, when a problem like this occurs, it is necessary to remove the cards from the robot and put them on the bench, as they are no longer seen by FirmwareUpdater, consequently, it is necessary to remove them and attach it to the test equipment to burn the bootloader again.