robotology/mesh-support #1318: Left and Right hand Fingertip Skin Stops Working When Under Stress

kouroshD, 2022-03-09:
### Device name 🤖

iCubGenova09 

### Request/Failure description

Currently, we are using the fingertip skin of the icub hand for telemanipulation.
**When we are manipulating object, i.e., the robot is under stress, some of the robot fingertip skins stops working. This specially happens very often with heavy object in the hand.**

### Detailed context

Here are the behavior of the received data:
- The tactile sensor values are high (saturated) and fixed
- The tactile sensor values are low/min and fixed
- Tactile senors blinks, jump between the low and high values
- Some times, all the skin does not work (show the previous behaviors), some time some of them behave like that, and other skins patches work fine
- Normally, when restarting again the `yarprobotinterface`  they start working again.

### Additional context

Here are an image showing the behavior:

[image]
This is the log file related to yesterdays experiment and photo.
[log_icub-head_yarprobotinterface_1403_skinCheckFingers.log](https://github.com/robotology/icub-tech-support/files/8213668/log_icub-head_yarprobotinterface_1403_skinCheckFingers.log)
In this experiment only two of the skin patches were working, and the rest were not working.

----
Here are some older logfiles when this happened:
[yarprunlog_16_02_2022_12_13_35.log](https://github.com/robotology/icub-tech-support/files/8213661/yarprunlog_16_02_2022_12_13_35.log)
[yarprunlog_16_02_2022_13_03_43.log](https://github.com/robotology/icub-tech-support/files/8213663/yarprunlog_16_02_2022_13_03_43.log)

cc @DanielePucci  @S-Dafarra

Uboldi80, 2022-03-11:
We have replaced 5 finertip in both hands: 

2 x L0
1 x R0
1 x R1
1 x L2

We test the robot qith @kouroshD and for now work well

closed 2022-03-11 by Uboldi80

kouroshD, 2022-03-11:
The fixes of yesterday are not relevant to this issue, I will open a ticket for this issue.
In any case, thank you @Uboldi80  for your support.