robotology/mesh-support #1319: [iCubGenova03] - left thumb distal tendon and fingertip

lindalastrico, 2022-03-09:
The left thumb tendons broke and the fingertip is not working

cc @Uboldi80 @gsisinna

Uboldi80, 2022-03-09:
As we discuss today, we cannot operate to fix this problems in the 16 March.
We need to replace a L0 fingertip and thumb distal 2391 cable.

Uboldi80, 2022-03-23:
Wednesday, March 30th we will be with you for the resolution of the problems described above.

cc. @GiuliaScorzaAzzara

gsisinna, 2022-03-30:
Today we restored the distal thumb tendon and the fingertip.

We found another problem with the skin of the right palm and therefore it will be necessary to schedule the next session with the researchers and prepare the necessary components.

cc @Uboldi @reafrancesco @GiuliaScorzaAzzara

closed 2022-03-30 by gsisinna