robotology/mesh-support #1320: Some of the Fingertip skins of the left and right hand does not work

kouroshD, 2022-03-09:
### Device name 🤖

iCubGenova09 

### Request/Failure description

These days, we are trying to perform the teloperation, and the robot skins are under some pressure.
Only a few of the skins on the left and right hand now are working as expected.

cc @DanielePucci  @S-Dafarra 

### Detailed context

Here is a video on the skin:

https://user-images.githubusercontent.com/17707730/157487405-8301c364-57b9-4399-b0f8-7db0c4f05a2b.mp4

Uboldi80, 2022-03-10:
Sorry guys 

but seem we have the same issue open in https://github.com/robotology/icub-tech-support/issues/1318

kouroshD, 2022-03-10:
@Uboldi80  They are different, in the sense that, when https://github.com/robotology/icub-tech-support/issues/1318 happens, after restarting the `yarprobotinterface`, those skins normally start to work again (HW/SW problem). Here, they are not working now, i.e., apparently HW problem.

kouroshD, 2022-03-10:
@Uboldi80  Please let me know when you have some time, we can investigate and work on this. Thanks

Uboldi80, 2022-03-10:
Today in the afternoon or Monday.

kouroshD, 2022-03-10:
@S-Dafarra  when do you suggest for this intervention?
@Uboldi80 this afternoon and Monday morning are fine for me. In any case, I have some meetings in between as well.

S-Dafarra, 2022-03-10:
The sooner the better :wink:

kouroshD, 2022-03-10:
@Uboldi80  when do you have time this afternoon?

Uboldi80, 2022-03-10:
14:30-17:00

kouroshD, 2022-03-10:
Sounds good to me. I have a meeting in between as well!

cc @S-Dafarra

Uboldi80, 2022-03-11:
I have replaced few fingertip as i described in the issu: https://github.com/robotology/icub-tech-support/issues/1318 and I fixed a short circuit in both hands:
1 short circuit in the wiring of the fingertip L0
1 short circuit in the mais right

closed 2022-03-11 by Uboldi80