robotology/mesh-support #1322: iCubGenova01 - Cannot detect sharedcan [1] joints 4 to 7

xEnVrE, 2022-03-14:
### Device name 🤖

iCubGenova01

### Request/Failure description

The robot startup fails as some boards are not reachable within the CAN network.

[image]

### Detailed context

The robot startup fails as some boards are not reachable within the CAN network.

[image]

gsisinna, 2022-03-16:
We investigated the problem more, and found several issues including:

- an MC4 has been replaced because it was faulty
- even a BLL board had to be replaced
- several connectors and crimps have been redone, including new CAN filters

Finally, there are also problems of short circuit and contact with cables in the palm for the part related to the pinky. For the latter it is necessary to schedule an intervention lasting at least 2 days and we will coordinate for this with @xEnVrE.

cc @Uboldi80 @Arya07 @maggia80

xEnVrE, 2022-03-18:
Given that the robot is not booked in the next days, we agreed with @Uboldi80 that the intervention could start today.

Uboldi80, 2022-03-23:
Finally this morning i tested the setup and all the problems seems to be solved.

closed 2022-03-23 by Uboldi80

xEnVrE, 2022-03-23:
Thanks @Uboldi80