robotology/mesh-support #1326: Right fingertip skin does not stream anymore data

GiulioRomualdi, 2022-03-15:
### Device name 🤖

iCubGenova09

### Request/Failure description

We started the teleoperation and all the fingertips were working then at the certain point we were able to retrieve the data only from the thumb and pinkie

cc @S-Dafarra @DanielePucci @Uboldi80 

### Detailed context

[log_icub-head_yarprobotinterface_6769_crc.txt](https://github.com/robotology/icub-tech-support/files/8254420/log_icub-head_yarprobotinterface_6769_crc.txt)

emilianob80, 2022-03-17:
@S-Dafarra The right fingertip electrical harness has been completely replaced and is now ready for testing. The bench fingertip was working so the present one was mounted

cc: @Uboldi80 @gsisinna @Fabrizio69

emilianob80, 2022-03-17:
A problem with MTB (2B7) of the wrist was found during fingertip restoration. It had to be replaced because the QuadroC connector was faulty