robotology/mesh-support #1327: Left shoulder roll and pitch goes in Hardware fault

GiulioRomualdi, 2022-03-15:
### Device name 🤖

iCubGenova09

### Request/Failure description

While teleoperating, the Left shoulder roll and pitch went in HF. We were not able to put the joints in idle but they went directly in HF.

We had to reboot the motors to restore a working configuration

cc @DanielePucci @S-Dafarra @Uboldi80 

### Detailed context

[log_icub-head_yarprobotinterface_6769_crc.txt](https://github.com/robotology/icub-tech-support/files/8254447/log_icub-head_yarprobotinterface_6769_crc.txt)

S-Dafarra, 2022-03-15:
This seems the same issue of https://github.com/robotology/icub-tech-support/issues/1321

cc @emilianob80 @Fabrizio69

S-Dafarra, 2022-03-16:
Let me stress that this does not happen at startup, but only after doing some motions with the shoulder

emilianob80, 2022-03-17:
After checking the wiring, the connectors of the ethernet lines and some CAN cables were redone. The problem is solved

closed 2022-03-17 by Uboldi80