robotology/mesh-support #1335: iCubGenova01 - Strange behavior of the left thumb and hand at startup

xEnVrE, 2022-03-29:
### Device name 🤖

iCubGenova01

### Request/Failure description

During the calibration procedure, the joint `hand_finger` started to oscillate visibly. After the calibration competed, the thumb opposition was as its maximum range and the thumb proximal and distal were moving erratically.

I am attaching a video:

https://user-images.githubusercontent.com/6014499/160661552-5468074f-3511-45ce-a72a-140c347aa6e7.mp4

After some seconds, part of the arms (from the wrist downwards) went in hardware fault.

I restarted the robot after this unlucky calibration and everything went smooth. Nonetheless, this strange behavior could have broken the thumb opposition, hence I think it is better to investigate why it happened.

Might it be that the new tactile sensors are making something like a short circuit?

### Detailed context

During the calibration procedure, the joint `hand_finger` started to oscillate visibly. After the calibration competed, the thumb opposition was as its maximum range and the thumb proximal and distal were moving erratically.

I am attaching a video:

https://user-images.githubusercontent.com/6014499/160661552-5468074f-3511-45ce-a72a-140c347aa6e7.mp4

After some seconds, part of the arms (from the wrist downwards) went in hardware fault.

I restarted the robot after this unlucky calibration and everything went smooth. Nonetheless, this strange behavior could have broken the thumb opposition, hence I think it is better to investigate why it happened.

Might it be that the new tactile sensors are making something like a short circuit?

xEnVrE, 2022-03-31:
I am attaching a log file which shows the error messages printed out when the above mentioned behavior happens.

The most relevant messages are:

```
9832,771038 <DEBUG> Can read/write buffers created, buffer size: 500
 9832,771052 <INFO> CanBusResources::initialized correctly
 9832,923180 <ERROR> recv buffer overrun ( 500 >=  500)
 9832,923309 <ERROR> run()-pushReadMsg() failed on CAN bus 1
 9832,923363 <ERROR> recv buffer overrun ( 500 >=  500)
 9832,923405 <ERROR> run()-pushReadMsg() failed on CAN bus 1
 9832,923444 <ERROR> recv buffer overrun ( 500 >=  500)
 9832,923482 <ERROR> run()-pushReadMsg() failed on CAN bus 1
 9832,923520 <ERROR> recv buffer overrun ( 500 >=  500)
 9832,923557 <ERROR> run()-pushReadMsg() failed on CAN bus 1
 9832,923593 <ERROR> recv buffer overrun ( 500 >=  500)
 9832,923599 <ERROR> run()-pushReadMsg() failed on CAN bus 1
 9832,923606 <ERROR> recv buffer overrun ( 500 >=  500)
 9832,933277 <ERROR> run()-pushReadMsg() failed on CAN bus 1
 9832,933316 <ERROR> recv buffer overrun ( 500 >=  500)
 9832,933326 <ERROR> run()-pushReadMsg() failed on CAN bus 1
 9832,933347 <ERROR> recv buffer overrun ( 500 >=  500)
 9832,933354 <ERROR> run()-pushReadMsg() failed on CAN bus 1
 9832,933361 <ERROR> recv buffer overrun ( 500 >=  500)
 9832,933367 <ERROR> run()-pushReadMsg() failed on CAN bus 1
 9832,933374 <ERROR> recv buffer overrun ( 500 >=  500)
 9832,933380 <ERROR> run()-pushReadMsg() failed on CAN bus 1
 9832,933387 <ERROR> recv buffer overrun ( 500 >=  500)
 9832,933393 <ERROR> run()-pushReadMsg() failed on CAN bus 1
 9832,933399 <ERROR> recv buffer overrun ( 500 >=  500)
 9832,933406 <ERROR> run()-pushReadMsg() failed on CAN bus 1
 9832,933412 <ERROR> recv buffer overrun ( 500 >=  500)
 9832,933419 <ERROR> run()-pushReadMsg() failed on CAN bus 1
 9832,933425 <ERROR> recv buffer overrun ( 500 >=  500)
 9832,933432 <ERROR> run()-pushReadMsg() failed on CAN bus 1
 9832,933438 <ERROR> recv buffer overrun ( 500 >=  500)
 9832,933444 <ERROR> run()-pushReadMsg() failed on CAN bus 1
 9832,933451 <ERROR> recv buffer overrun ( 500 >=  500)
 9832,933457 <ERROR> run()-pushReadMsg() failed on CAN bus 1
 9832,933464 <ERROR> recv buffer overrun ( 500 >=  500)
 9832,933470 <ERROR> run()-pushReadMsg() failed on CAN bus 1
 9832,933476 <ERROR> recv buffer overrun ( 500 >=  500)
 9832,933490 <ERROR> run()-pushReadMsg() failed on CAN bus 1
(...) and continues like that
```

The messages above appear later also for the `CAN bus 2`.

Also, one can find the following:

```
9870,298474 <INFO> icub/cartesianController/right_arm: Checking if cartesian solver cartesianSolver/right_arm is alive... not yet
 9871,058052 <INFO> icub/cartesianController/left_arm: Checking if cartesian solver cartesianSolver/left_arm is alive... not yet
 9871,316133 <INFO> icub/cartesianController/right_arm: Checking if cartesian solver cartesianSolver/right_arm is alive... not yet
 9872,082651 <INFO> icub/cartesianController/left_arm: Checking if cartesian solver cartesianSolver/left_arm is alive... not yet
 9872,289457 <ERROR> sharedcan [2] board 3 FAULT UNDERVOLTAGE AXIS 0
 9872,289589 <ERROR> sharedcan [2] board 3 FAULT UNDERVOLTAGE AXIS 1
 9872,289663 <ERROR> sharedcan [2] board 4 FAULT UNDERVOLTAGE AXIS 0
 9872,299609 <ERROR> sharedcan [2] board 4 FAULT UNDERVOLTAGE AXIS 1
 9872,299743 <ERROR> sharedcan [1] board 3 FAULT UNDERVOLTAGE AXIS 0
 9872,299796 <ERROR> sharedcan [1] board 3 FAULT UNDERVOLTAGE AXIS 1
 9872,299837 <ERROR> sharedcan [1] board 4 FAULT UNDERVOLTAGE AXIS 0
 9872,309709 <ERROR> sharedcan [1] board 4 FAULT UNDERVOLTAGE AXIS 1
 9872,309841 <ERROR> sharedcan [5] board 7 FAULT UNDERVOLTAGE AXIS 0
 9872,309895 <ERROR> sharedcan [5] board 7 FAULT UNDERVOLTAGE AXIS 1
 9872,309938 <ERROR> sharedcan [5] board 8 FAULT UNDERVOLTAGE AXIS 0
 9872,309978 <ERROR> sharedcan [5] board 8 FAULT UNDERVOLTAGE AXIS 1
 9872,330308 <INFO> icub/cartesianController/right_arm: Checking if cartesian solver cartesianSolver/right_arm is alive... not yet
 9872,412457 <ERROR> AnalogWrapper: AnalogServer: Sensor returned timeout error (code 3).
 9872,422837 <ERROR> AnalogWrapper: AnalogServer: Sensor returned timeout error (code 3).
 9872,423022 <ERROR> AnalogWrapper: AnalogServer: Sensor returned timeout error (code 3).
 9872,433137 <ERROR> AnalogWrapper: AnalogServer: Sensor returned timeout error (code 3).
 9872,433292 <ERROR> AnalogWrapper: AnalogServer: Sensor returned timeout error (code 3).
 9872,443232 <ERROR> AnalogWrapper: AnalogServer: Sensor returned timeout error (code 3).
 9872,443415 <ERROR> AnalogWrapper: AnalogServer: Sensor returned timeout error (code 3).
(...) and continues like that
```

[log_icub-head_yarprobotinterface_1132.txt](https://github.com/robotology/icub-tech-support/files/8391478/log_icub-head_yarprobotinterface_1132.txt)

Uboldi80, 2022-04-01:
Today i solved the problem with the MTB, but that is another problem not correlated to the thumb movement.
The drift of the thumb is generated by a short circuit trough the patch cable LTP that pushed on the FTC of some finger.
I have fix some short circuit on the FTC and insulate by the glue.
now  seems to work well.
@xEnVrE let me know and if ok, please close the issue.

xEnVrE, 2022-04-01:
Seems to be solved now, closing

closed 2022-04-01 by xEnVrE