robotology/mesh-support #1336: Short circuit in the right hand wiring

Uboldi80, 2022-03-30:
Today we notced a problem during the test of setup.
We find a short circuit inside the cable that connect the FTC BOARD and the MTB in the forearm.

This problem causes spikes when using the right hand skin.

cc. @GiuliaScorzaAzzara   @lindalastrico  @reafrancesco

gsisinna, 2022-03-31:
Maintenance for iCubGenova03 will be scheduled on **April 13**.

@GiuliaScorzaAzzara let us know if it works for you and your team, thanks!

cc @Uboldi80

gsisinna, 2022-04-13:
@Uboldi80 replaced the RTP cable patch (right_hand) but we noticed that the whole hand palm skin patch still not working. We're closing this issue and opening an other for the next session.

cc @lindalastrico @GiuliaScorzaAzzara

closed 2022-04-13 by gsisinna