robotology/mesh-support #1337: iCubGenova01 - Left thumb distal moves correctly only in one direction

xEnVrE, 2022-04-01:
### Device name 🤖

iCubGenova01

### Request/Failure description

After the calibration, the left thumb distal stays in the following configuration:

[image]

If I try to move it from the `yarpmotorgui` the finger closes but does not open.

### Detailed context

After the calibration, the left thumb distal stays in the following configuration:

[image]

If I try to move it from the `yarpmotorgui` the finger closes but does not open.

Uboldi80, 2022-04-01:
I noticed that between the curtain and the pulley near the motor there was dirt that tended to rub the cable preventing it from moving freely.
I removed what I found and cleaned out the venue.

xEnVrE, 2022-04-01:
Thanks @Uboldi80 

Closing as the issue has been solved

closed 2022-04-01 by xEnVrE