robotology/mesh-support #1341: Some fingertips are not working

S-Dafarra, 2022-04-01:
### Device name 🤖

iCubGenova09

### Request/Failure description

During the Xprize semifinals, we noticed that the left thumb and right pinky fingertips do not work.

### Detailed context

It seems that the supports are broken too

emilianob80, 2022-04-04:
@S-Dafarra After removing the back of the hand on both hands there were 2 FTC connectors partially removed from the board. These 2 corresponded to the non-functioning fingertips. A functional test will be carried out to verify the effective functioning of the fingertips

cc: @gsisinna @Uboldi80 @Fabrizio69

emilianob80, 2022-04-04:
Unfortunately from the functional tests the non-working fingertips are:
right thumb tip
left little finger
left middle fingertip

emilianob80, 2022-04-04:
@S-Dafarra  it was necessary to replace the thumb and index finger of the right hand and the little finger of the left hand. On the left hand the interruption of the orange wire of the cables that connected the tip of the little finger was found. At the moment, for reasons of timing and not of urgency, the cable braid has not been replaced but repaired (see photo below).

[image]

All fingertips function properly

I open a warehouse [ticket](https://github.com/icub-tech-iit/tickets/issues/2164) to unload the fingertips used during the intervention

cc: @Uboldi80 @gsisinna @Fabrizio69

closed 2022-04-04 by gsisinna