robotology/mesh-support #1344: [iCubGenova07] - Right thumb broken due to self collision

GiuliaScorzaAzzara, 2022-04-05:
### Device name 🤖

iCubGenova07

### Request/Failure description

During the execution of a demo the robot self collided causing the break of the oppose joint in the right thumb.

### Detailed context

[image]

gsisinna, 2022-04-11:
Hi @GiuliaScorzaAzzara,
we can schedule the maintenance for this issue on Wednesday 13th (this week). 

Is the robot available?

Thanks!

cc @Uboldi80

GiuliaScorzaAzzara, 2022-04-11:
Hi @gsisinna,
It is available in the morning, thanks! 
See you on wednesday!

gsisinna, 2022-04-13:
The thumb has been restored and here there are the new configuration files for the right_arm:
[iCubGenova07.zip](https://github.com/robotology/icub-tech-support/files/8480273/iCubGenova07.zip)

I will make a pull request after a general check of the files online that are misaligned with respect to the local ones.

cc @Uboldi80 @GiuliaScorzaAzzara

closed 2022-04-13 by gsisinna