robotology/mesh-support #1345: [iCubGenova04] robot does not start because of problems on the right leg ankle

gabrielenava, 2022-04-05:
### Device name 🤖

iCubGenova04

### Request/Failure description

We were performing the joints fine calibration with the robot on the pole, and at a certain point the robot did not start anymore. It seems there is a problem with the right leg ankle board:

```
[ERROR] EthResource::verifyBoardPresence() DID NOT have replies from BOARD right_leg-eb9-j4_5 with IP 10.0.1.9 after 21.0061 seconds: CANNOT PROCEED ANY FURTHER
[ERROR] EthResource::verifyEPprotocol() cannot verify BOARD right_leg-eb9-j4_5 with IP 10.0.1.9 : cannot proceed any further
[ERROR] embObjMotionControl: failed verifyEPprotocol. Cannot continue!
[ERROR] theNVmanager::Impl::wait() had a timeout for BOARD right_leg-eb9-j4_5 IP 10.0.1.9 and nv ID32 = 0x00030001 -> IND = 0, TAG = eoprot_tag_mn_service_status_commandresult
[ERROR] theNVmanager::Impl::command() fails a wait() from IP 10.0.1.9 for nv ID32 = 0x00030001 -> IND = 0, TAG = eoprot_tag_mn_service_status_commandresult
[ERROR] theNVmanager::Impl::wait() had a timeout for BOARD right_leg-eb9-j4_5 IP 10.0.1.9 and nv ID32 = 0x00030001 -> IND = 0, TAG = eoprot_tag_mn_service_status_commandresult
[ERROR] theNVmanager::Impl::command() fails a wait() from IP 10.0.1.9 for nv ID32 = 0x00030001 -> IND = 0, TAG = eoprot_tag_mn_service_status_commandresult
[ERROR] theNVmanager::Impl::wait() had a timeout for BOARD right_leg-eb9-j4_5 IP 10.0.1.9 and nv ID32 = 0x00030001 -> IND = 0, TAG = eoprot_tag_mn_service_status_commandresult
[ERROR] theNVmanager::Impl::command() fails a wait() from IP 10.0.1.9 for nv ID32 = 0x00030001 -> IND = 0, TAG = eoprot_tag_mn_service_status_commandresult
[ERROR] EthResource::serviceCommand() failed an acked activation request to BOARD right_leg-eb9-j4_5 with IP 10.0.1.9 after 3 attempts each with waiting timeout of 0.5 seconds
[ERROR] |yarp.dev.PolyDriver| Driver <embObjMotionControl> was found but could not open
[WARNING] Cannot open device right_leg-eb9-j4_5-mc
[WARNING] Cannot open device right_leg-eb9-j4_5-mc
[ERROR] |yarp.devices.ControlBoardWrapper| Parameter networks use deprecated syntax
[ERROR] |yarp.devices.ControlBoardWrapper| Parameter networks use deprecated syntax

```

### Detailed context

here the loggers:

[right_ankle_failure_all_attempts.txt](https://github.com/robotology/icub-tech-support/files/8420429/right_ankle_failure_all_attempts.txt)
[right_ankle_failure_first_attempt.txt](https://github.com/robotology/icub-tech-support/files/8420430/right_ankle_failure_first_attempt.txt)

Gandoo, 2022-04-07:
replaced E11 cable from Ems EB11 to EB9

[image]

it looks throttled and damaged in this position

[image]

cc @gabrielenava @Uboldi80 @gsisinna

GiacomoVassallo13, 2022-04-07:
Buongiorno cavo 14 della gamba destra ripristinato con successo.

cc: @gabrielenava @gsisinna @Gandoo @Uboldi80

HosameldinMohamed, 2022-04-07:
> Buongiorno cavo 14 della gamba destra ripristinato con successo.

Hi @GiacomoVassallo13 Bad news

I just started the robot and then moved the right leg pitch with the GUI, the cable went out!
[image]

GiacomoVassallo13, 2022-04-07:
the cable problem has been solved. unfortunately, however, the back cover remained damaged. The piece already showed signs of breakage and glue.

cc: @gabrielenava @Gandoo @Fabrizio69

HosameldinMohamed, 2022-04-07:
Thanks @GiacomoVassallo13 , I tested the right leg pitch with `yarpmotorgui` and it seems OK!

But I see an issue with the `torso` board. The joints are always in hardware fault..

I think the relevant message from the logger is:

```
MC: AEA encoder invalid data. Hardware problem in the magnetic position sensor of the joint caused invalid position readings. par16 = AEA port (msb) and ID of joint (lsb). +
```

Here's the whole log
[yarprunlog_07_04_2022_18_01_07.log](https://github.com/robotology/icub-tech-support/files/8444806/yarprunlog_07_04_2022_18_01_07.log)

HosameldinMohamed, 2022-04-07:
@GiacomoVassallo13 The leg cable is out again! 😞

gabrielenava, 2022-04-07:
I have changed the issue title in view of the multiple problems that emerged while working on the original issue. A summary:

- we requested support for the right ankle board that was not able to communicate. See https://github.com/robotology/icub-tech-support/issues/1345#issue-1193482496

- with a fast visual inspection we discovered there was a faulty cable; the problem was quickly solved by @Gandoo in https://github.com/robotology/icub-tech-support/issues/1345#issuecomment-1091621996

- while the robot was in the facility for the problem on the ankle, @Gandoo and @GiacomoVassallo13 noted that left leg hip tendon was damaged too, and proceeded replacing it, see https://github.com/robotology/icub-tech-support/issues/1345#issuecomment-1091764869.
 Problems on the right hip motor have been noted few days ago during the [checkup after the wind tunnel experiment](https://github.com/icub-tech-iit/tickets/issues/2143#issuecomment-1083005770) and the leg had been disassembled and mounted back to check the status of the hip motors. Since then, the robot has not been used, besides turning it on on the pole for sensors calibration.

- unfortunately, for an unknown reason, the cable keeps exiting the guide. Furthermore, while working on the robot for fixing the hip problem, another issue that was not there before appeared: https://github.com/robotology/icub-tech-support/issues/1345#issuecomment-1091940924. The new problem occurred in the same area of the robot where we were working for fixing the hip issue, so the two may be related.

at this point we have no choice but to postpone the restart of iRonCub experiment to after Easter @DanielePucci 

cc @maggia80 @Fabrizio69

GiacomoVassallo13, 2022-04-11:
@gabrielenava the cable 14 of the right leg has been restored again. In the previous cases the cable tensioner was mounted or on the contrary, now that it has been restored, the problem should no longer occur. after this series of restorations it is necessary to recalibrate the right leg to avoid inconveniences

CC: @Fabrizio69 @Gandoo

gsisinna, 2022-04-12:
Yesterday together with the team guys we noticed that the joint that gives problems with the torso is the "torso roll". The AEA reading does not change when the position of the joint changes, and I tried to replace the AEA sensor but the result is the same.

It is necessary to go and see in the EMS (EB5 board of the waist) if some connector is damaged or some wire is broken/bent.

cc @gabrielenava @Gandoo @emilianob80 @Uboldi80

Uboldi80, 2022-04-14:
Today with @gsisinna  we completed the operation, replacing the EMS4 EB5 which had the P6 TORSO ROLL channel not working, probably due to a helicoil that had fallen from its seat to end up in a short circuit on the EMS4.

closed 2022-04-14 by Uboldi80