robotology/mesh-support #1347: An error occurred during iCub software installation.

SSSSamZhu, 2022-04-11:
### Device name 🤖

iCubShenzhenS/N0.39

### Request/Failure description

We plan to reinstall iCub software on a new workstation to back up the iCub system.However, after reinstalling the software, we found that the error as shown in the figure would be reported when running the` yarpmanager`command. Meanwhile, the `iCub_SIM` command could not be found.

The installation method we use is this https://icub-tech-iit.github.io/documentation/sw_installation/linux_from_sources_manual/#setup-icubcontrib-optional

Or should I repeat the installation steps?I've only been with iCub for two weeks，thanks.

### Detailed context

[image]

gsisinna, 2022-04-11:
Hi @SSSSamZhu,
thanks for writing. So, as a first thing I would try to figure out with you what version of the software you are using on the robot and how you intend to install the rest on the laptop.

From what I can understand from the picture you are not using the most recent installation procedure based on the robotology-superbuild, but you have followed the procedure where it is necessary to compile each module/component of the ecosystem for iCub (YARP, YCM, icub-main) individually.

To keep in line with this last method I suggest you to refer to our official guide in which are listed the various steps for a manual compilation of the single packages: [Install iCub software from source manually installing each software package](https://icub-tech-iit.github.io/documentation/sw_installation/linux_from_sources_manual/)

For any other doubt we remain at your disposal!

SSSSamZhu, 2022-04-13:
Hi @gsisinna ，
Thank you for your reply.I have operated it again according to the online document these days.I recompiled YARP and solved some errors reported before, but the problem still exists.Below is a screenshot of the latest error.

[image]

version of the software I chose
Ubuntu 20.04 YARP v3.6.1 iCub v1.24.0

pattacini, 2022-04-13:
Hi @SSSSamZhu 

It seems that you simply stopped `yarpserver` right before launching `yarpmanager`, which in turn complained about that.

You could try launching `yarpmanager` in a second console then, while keeping `yarpserver` running in the first console.

gsisinna, 2022-04-13:
@SSSSamZhu in addition to what @pattacini correctly said, you need to check that all clusters and their names are set correctly in yarpmanager. You can check if this file exists in the following path:  `/home/icub/.local/share/yarp/contexts`

```
<cluster name="icub01" user="icub">
<nameserver namespace="/root" node="**icubsrv**"> </nameserver>
<!-- specify nodes, if you want to enable display on desktop please specify which desktop to use
this is usually the value of the DISPLAY variable when logged on -->
<node display=":0.0">icubsrv</node>
<node>icub-head</node>
</cluster>
```

You must insert the node name where you want to launch the yarp nameserver (in my case icubsrv).

SSSSamZhu, 2022-04-13:
Thank you for your reply.In the past few days, the wireless network card of the workstation was broken, so I took photos, and I will use screenshots later.

I have just finished the SSH configuration of iCub, but I did not see the file mentioned by@gsisinna. May I ask if there is a link to the corresponding configuration file.

gsisinna, 2022-04-13:
@SSSSamZhu does this folder exist regardless of the file?

`/home/icub/.local/share/yarp/contexts`

Thanks!

pattacini, 2022-04-13:
If `/home/icub/.local/share/yarp/contexts/yarpmanager` is not present, you could create it by launching:

```console
yarp-config context --import yarpmanager
```

SSSSamZhu, 2022-04-15:
Hi @gsisinna ，
I'm glad to tell you that I've finished installing icub,but there are still a few minor issues to work out.After entering Yarpmanager,I could not find yarpstartup and other options in Applications.
[image]
[image]

gsisinna, 2022-04-15:
Hi @SSSSamZhu you can try with this: [entities-yarpmanager.zip](https://github.com/robotology/icub-tech-support/files/8495027/entities-yarpmanager.zip)

This files needs to be copied into the laptop server after compilation, specifically here:

`/home/icub/.local/share`

The folder contains all the applications and contexts you mentioned.

When running yarpmanager, ensure that all the modules are listed in the correct order.

SSSSamZhu, 2022-04-18:
Hi @gsisinna ，
I have copied the above files to the appropriate path and modified all the 'icubsrv' in the.xml file.(I had to redefine the host name because of the network access requirements of the company computer)
But after I complete these operations, I can only complete operations unrelated to "ICubSRV" in yarpManager interface, as shown in the picture "YarprobotInterface".
[image]
[image]

Is there anything else I need to edit if I want to continue using this host name?

gsisinna, 2022-04-19:
Hi @SSSSamZhu, can you show me your **Cluster** tab and the logs from the terminal where you launched yarpmanager?

Thanks!

SSSSamZhu, 2022-04-19:
Hi @gsisinna ，

Cluster tab 
[image]

the logs from yarpmanager
`[INFO] |yarp.os.Port|/tmp/port/1| Port /tmp/port/1 active at tcp://192.168.1.100:10005/
[INFO] |yarp.os.impl.PortCoreOutputUnit|/tmp/port/1| Sending output from /tmp/port/1 to /isaacliang-LC3 using tcp
[INFO] |yarp.os.impl.PortCoreOutputUnit|/tmp/port/1| Removing output from /tmp/port/1 to /isaacliang-LC3
[INFO] |yarp.os.Port|/tmp/port/1| Port /tmp/port/1 active at tcp://192.168.1.100:10005/
[INFO] |yarp.os.impl.PortCoreOutputUnit|/tmp/port/1| Sending output from /tmp/port/1 to /isaacliang-LC3 using tcp
[INFO] |yarp.os.impl.PortCoreOutputUnit|/tmp/port/1| Removing output from /tmp/port/1 to /isaacliang-LC3
[INFO] |yarp.os.Port|/tmp/port/1| Port /tmp/port/1 active at tcp://192.168.1.100:10005/
[INFO] |yarp.os.impl.PortCoreOutputUnit|/tmp/port/1| Sending output from /tmp/port/1 to /isaacliang-LC3 using tcp
[INFO] |yarp.os.impl.PortCoreOutputUnit|/tmp/port/1| Removing output from /tmp/port/1 to /isaacliang-LC3
`
 the file /etc/hosts
[image]

gsisinna, 2022-04-19:
Thanks @SSSSamZhu, can you try to execute this commands in a shell?

`ssh-keygen -t rsa`

`ssh-copy-id -i /home/icub/.ssh/id_rsa.pub icub@isaacliang-LC3`

I think you still have some problems with yarpserver, yarprun and the cluster nodes connections.

After that you can try:
1. clean restart and launch yarpserver from yarpmanager-GUI
2. connect the cluster nodes
3. see the logs file

gsisinna, 2022-04-19:
@SSSSamZhu if you want we can schedule also a videocall to help you with this issues, let me know!

SSSSamZhu, 2022-04-19:
Hi @gsisinna ，
the logs file after executing this commands
[image]

[image]

gsisinna, 2022-04-19:
This afternoon I'll be available. Is it ok for you?

I can share you the invite if you send me your email address, thanks!

SSSSamZhu, 2022-04-19:
Hi，@gsisinna Maybe we can arrange the video at 3:00-5:00pm Beijing time (9:00-11:00am Italian time).  My email address：13702661257z@gmail.com

gsisinna, 2022-04-19:
@SSSSamZhu do you have an institutional mail address? For security reasons, I cannot send requests to such an email from my institution.

I can schedule for tomorrow from 11.30am to 12.30am or on April 22nd from 9.30am to 11.00am.

Thanks

SSSSamZhu, 2022-04-19:
@gsisinna how about yi_liu@mail2.gdut.edu.cn 
Thanks

gsisinna, 2022-04-19:
Invite sent.

Thanks,
Gabriele 

cc @Uboldi80

SSSSamZhu, 2022-04-20:
Actually I saw the configuration for PC104 when I was looking through the documentation today.
https://icub-tech-iit.github.io/documentation/icub_operating_systems/pc104/the-icub-live-details/

`127.0.0.1   localhost
127.0.1.1   pc104.icub.local pc104
10.0.0.1    icubsrv
`
Is icubsrv here a default value?Do I need to modify the files on PC104?

gsisinna, 2022-04-20:
Hi @SSSSamZhu , 
as I have already explained in the call for now leave the configuration files in this way and as soon as possible we will try to organize a call with the control of your screen.

We have tried to copy the **ssh** key since you changed the hostname after having already saved this key in the registry, and now it seems that the laptop cannot establish an SSH connection on itself to run **yarprun** and start the applications.

**icubsrv** can be set as an alias for your laptop, but in this case you should refer to your **isaacliang-LC3** as hostname.

This is a new workstation, so is the laptop that came with the robot working properly?

For the next call, could we have the support of someone who speaks English and can follow along with the instructions?

Thanks!

cc @Uboldi80

SSSSamZhu, 2022-04-21:
Hi，@gsisinna ，
Tthe laptop that came with the robot is working properly.I'll ask a colleague to assist me next time we meet.
Thanks

gsisinna, 2022-04-21:
Thanks @SSSSamZhu,
consider that when installing a new software it is necessary to follow in detail the official procedure in order to match users, privileges, hostname, aliases and environment variables. So, I ask you to check that all the basic setup is aligned before scheduling the next call, referring to:

1. [Generic iCub environment machines](https://icub-tech-iit.github.io/documentation/icub_operating_systems/other-machines/generic-machine/)
2. [The iCub Console Server](https://icub-tech-iit.github.io/documentation/icub_operating_systems/other-machines/icub-server-laptop/)
3. [Install iCub software from source using the robotology-superbuild on Linux](https://icub-tech-iit.github.io/documentation/sw_installation/linux_from_sources_superbuild/)

Talk soon!

SSSSamZhu, 2022-04-24:
Hi，@gsisinna ，
I reconfigured my computer in the last two days, starting from [here](https://icub-tech-iit.github.io/documentation/icub_ope) and going all the way to SSH configuration,We regenerated the SSH key.But unfortunately, icub still has the usual problems in the startup.He has been unable to perform tasks related to isaac Liang-LC3 node on yarpmanager, but he can perform tasks related to PC104 node.
Looking forward to your reply.

SSSSamZhu, 2022-04-24:
Hi，@gsisinna，
At present, I have solved my problem and all applications of icub can work normally for the time being.I modified the hosts file on PC104 and set the icubsrv alias.
Thanks for your help,I will close this issue.

closed 2022-04-24 by SSSSamZhu