robotology/mesh-support #1349: iCubGenova09 left shoulder 2FOC recurrent problem

S-Dafarra, 2022-04-11:
### Device name 🤖

iCubGenova09

### Request/Failure description

The 2FOC controlling the left should pitch and roll sometimes stops communicating with the EMS. This problem occurred twice in the past, and it happened again while preparing the demo at the Ministero della Salute.

### Detailed context

The issue is very similar to:
- https://github.com/robotology/icub-tech-support/issues/1327
- https://github.com/robotology/icub-tech-support/issues/1321

In particular, the logs contain errors like
```
10341,604664 <ERROR> from BOARD 10.0.1.1 (left_arm-eb1-j0_1), src CAN1, adr 0, time 2403s 113m 129u: (code 0x00000019, par16 0x0101 par64 0x0000000000000000) -> SYS: EOtheCANservice could not tx frames on CAN bus. In par16 there is: on msb the size of txfifo, on lsb a code. + .
10341,614459 <WARNING> from BOARD 10.0.1.1 (left_arm-eb1-j0_1), src LOCAL, adr 0, time 2403s 113m 285u: (code 0x0000000d, par16 0x017e par64 0x000100260014004a) -> SYS: the TX phase of the control loop has last more than wanted. In par16: TX execution time [usec]. In par64: num of tx can2 and can1 frames and latest previous execution times of TX, RX, DO [usec] + .
10341,614820 <ERROR> from BOARD 10.0.1.1 (left_arm-eb1-j0_1), src CAN1, adr 0, time 2403s 116m 129u: (code 0x00000019, par16 0x0101 par64 0x0000000000000000) -> SYS: EOtheCANservice could not tx frames on CAN bus. In par16 there is: on msb the size of txfifo, on lsb a code. + .
10341,624835 <WARNING> from BOARD 10.0.1.1 (left_arm-eb1-j0_1), src LOCAL, adr 0, time 2403s 116m 287u: (code 0x0000000d, par16 0x0180 par64 0x0001017e00ff039e) -> SYS: the TX phase of the control loop has last more than wanted. In par16: TX execution time [usec]. In par64: num of tx can2 and can1 frames and latest previous execution times of TX, RX, DO [usec] + .
10341,634730 <ERROR> from BOARD 10.0.1.1 (left_arm-eb1-j0_1), src CAN1, adr 0, time 2403s 119m 129u: (code 0x00000019, par16 0x0201 par64 0x0000000000000000) -> SYS: EOtheCANservice could not tx frames on CAN bus. In par16 there is: on msb the size of txfifo, on lsb a code. + .
```

This seems to indicate that the EMS stops communicating with the 2FOC. @emilianob80 checked several times all the connectors, also rewiring everything. Right before the semifinals, he also changed the 2FOC. Unfortunately though, the problem occurred again while preparing the demo at the Ministero della Salute.

### Additional context

An important thing to mention is that, once this happens, it is not possible to recover. On the other hand, it is often enough to reboot the motors to start working again.

This puzzles me a bit, because if the problem is in the connection, I wonder why it is enough to reboot to make it working again. Hence, I wonder if the problem is on the wiring (although it has been checked), on the 2FOC (although it has been already substituted), or in the EMS.

cc @maggia80 @pattacini @marcoaccame @isorrentino @lrapetti @mebbaid @gsisinna

S-Dafarra, 2022-04-11:
Here the log file from the failure during the tests at Ministero della Salute 
[log_icub-head_yarprobotinterface_4711_hf_left_shoulder.txt](https://github.com/robotology/icub-tech-support/files/8466386/log_icub-head_yarprobotinterface_4711_hf_left_shoulder.txt)

Uboldi80, 2022-04-15:
we tested the robot and it seems should be ok!

closed 2022-04-15 by Uboldi80