robotology/mesh-support #1350: Robot unable to start, error on left leg CAN

lrapetti, 2022-04-12:
### Device name 🤖

iCubGenova09

### Request/Failure description

The robot can not start, see the attached log file

### Detailed context

We experienced an overcurrent on hip pitch during the usage. After restarting the robot, the robotinterface is not starting.
The error is the following (full log file below)
```
 77,924141 <WARNING>  from BOARD 10.0.1.7 (left_leg-eb7-j0_2), src LOCAL, adr 0, time 52s 514m 276u: (code 0x0000000b, par16 0x01e1 par64 0x019701e1008b0197) -> SYS: the RX phase of the control loop has last more than wanted. In par16: RX execution time [usec]. In par64: latest previous execution times of TX, RX, DO, TX [usec] + .
 77,924250 <ERROR>  from BOARD 10.0.1.7 (left_leg-eb7-j0_2), src LOCAL, adr 0, time 52s 550m 429u: (code 0x00000013, par16 0x0001 par64 0x0000000000000000) -> SYS: the EOtheInfoDispatcher could not accept a eOmn_info_properties_t item inside its transmitting queue. In par16 there is the number of lost items. + .
```
[log_icub-head_yarprobotinterface_LeftLeg_CAN.txt](https://github.com/robotology/icub-tech-support/files/8471979/log_icub-head_yarprobotinterface_LeftLeg_CAN.txt)

lrapetti, 2022-04-12:
cc @mebbaid @S-Dafarra @gsisinna @Uboldi80

S-Dafarra, 2022-04-12:
Can you check from ``FirmwareUpdater`` which boards can be detected from the 10.0.1.7 board?

S-Dafarra, 2022-04-12:
Also this error seems juicy: 
```
77,793606 <ERROR>  from BOARD 10.0.1.7 (left_leg-eb7-j0_2), src CAN2, adr 0, time 52s 391m 126u: (code 0x00000019, par16 0x4001 par64 0x0000000000000000) -> SYS: EOtheCANservice could not tx frames on CAN bus. In par16 there is: on msb the size of txfifo, on lsb a code. + .
```
I guess it means that it is not able to send packets on the CAN2. By looking at https://icub-tech-iit.github.io/documentation/icub_wiring/icub3_x/#icub-3x-wiring, it seems that a foot FT is connected to the CAN2. Maybe a cable got disconnected

mebbaid, 2022-04-12:
> Can you check from `FirmwareUpdater` which boards can be detected from the 10.0.1.7 board?

Here is the result of running the `FirmwareUpdater` in the head and forcing the `10.0.1.7` ems to maintainance
[image]

S-Dafarra, 2022-04-12:
From https://github.com/ami-iit/robots-configuration/blob/d36d1fd3bee49cbe7d47f1b651f35d2c5979a75e/iCubGenova09/hardware/FT/left_leg-eb7-j0_2-strain.xml it seems that the FT connected to the eb7 is the left heel FT.

mebbaid, 2022-04-12:
The disconnected cable is not visible to me from by just looking at the heel of the robot @lrapetti  @S-Dafarra @gsisinna

gsisinna, 2022-04-12:
Seeing the FirmwareUpdater figure you can see a total disconnection of the hardware from the board. Soon I or @Uboldi80 will check all the connections necessary to restore the communication.

lrapetti, 2022-04-13:
> Seeing the FirmwareUpdater figure you can see a total disconnection of the hardware from the board. Soon I or @Uboldi80 will check all the connections necessary to restore the communication.

Thanks a lot @gsisinna!

Note that this seems to be the only blocking issue for experimental activities on `iCubGenova09`, so if possible I would give priority with respect to other problems on this robot (https://github.com/robotology/icub-tech-support/issues/1348, https://github.com/robotology/icub-tech-support/issues/1349) for which we can coordinate later on (and might require more effort).
cc @DanielePucci @S-Dafarra

S-Dafarra, 2022-04-14:
So the problem was not in the FT right?

Uboldi80, 2022-04-14:
I also checked the FT cable but it doesn't seem to have any obvious problems

Uboldi80, 2022-04-15:
we tested the robot and it seems the problem is solved!

closed 2022-04-15 by Uboldi80