robotology/mesh-support #1354: The 0B3 2FOC in the torso gets really hot while walking

S-Dafarra, 2022-04-19:
### Device name 🤖

iCubGenova09

### Request/Failure description

@gsisinna noticed that the temperature of the 0B3 2FOC located in the torso can get very hot. 

### Detailed context

Interestingly, the corresponding joints are not very used during walking. This board is also attached to the EMS throwing CRC errors (https://github.com/robotology/icub-tech-support/issues/1195)

gsisinna, 2022-04-19:
The board should be ready to mount an additional heat sink.

[image]

cc @maggia80 @Uboldi80

maggia80, 2022-04-20:
@S-Dafarra do you have current measurements of these 3 joints during an experiment?

S-Dafarra, 2022-04-20:
Here some logs https://istitutoitalianotecnologia.sharepoint.com/:f:/s/DynamicInteractionControl/EkN8h5iJ9mxBqFdvf4EHHZYB-_5NFt3kfLUSbwOZO9ljUw?e=L7LMcL with the associated video. Let me know if you need support with them

gsisinna, 2022-04-26:
From the 2FOC board documentation I can read:

> 4.2.2.3 Over-current protection
> The currents in the bridge exceeding 21A are detected by a fast comparator (U19) and a shunt resistor (R15) and generate the overload signal OVL that, without firmware intervention, stops the PWM generators to avoid damaging the electronics.
> The power stage relies on an external high value capacitor (2200μF or more) connected to the DC-Link rails in order to absorb the over voltage generated during motor braking.

Addiotional resources: [iCub3 Waist Pitch joint](https://icub-tech-iit.github.io/documentation/icub3/icub3-joints/#joint-5-waist-pitch)

[Custom components (motors)](https://icub-tech-iit.github.io/documentation/custom_components/custom_components/)

We should study in detail the current drawn by the motor during the walking phase at the end of the day before the demo, after several hours of heavy use. 

@GiulioRomualdi can you send me the joints mapping for the robot-log-visualizer? I can see only progressive numbers.

[image]

Uboldi80, 2022-04-27:
Today, me and @Elio75 arranged a heatsink to add on the 2FOC 0B3 to try to increase its dissipation capacity.
[image]
[image]

traversaro, 2022-04-27:
> @GiulioRomualdi can you send me the joints mapping for the robot-log-visualizer? I can see only progressive numbers.

If you open the dataset in MATLAB or some other environment that supports `.mat` files, the `robot_logger_device.description_list` array should give you the list:
~~~
>> robot_logger_device.description_list

ans =

  32×1 cell array

    {'neck_pitch'      }
    {'neck_roll'       }
    {'neck_yaw'        }
    {'torso_pitch'     }
    {'torso_roll'      }
    {'torso_yaw'       }
    {'l_shoulder_pitch'}
    {'l_shoulder_roll' }
    {'l_shoulder_yaw'  }
    {'l_elbow'         }
    {'l_wrist_prosup'  }
    {'l_wrist_pitch'   }
    {'l_wrist_yaw'     }
    {'r_shoulder_pitch'}
    {'r_shoulder_roll' }
    {'r_shoulder_yaw'  }
    {'r_elbow'         }
    {'r_wrist_prosup'  }
    {'r_wrist_pitch'   }
    {'r_wrist_yaw'     }
    {'l_hip_pitch'     }
    {'l_hip_roll'      }
    {'l_hip_yaw'       }
    {'l_knee'          }
    {'l_ankle_pitch'   }
    {'l_ankle_roll'    }
    {'r_hip_pitch'     }
    {'r_hip_roll'      }
    {'r_hip_yaw'       }
    {'r_knee'          }
    {'r_ankle_pitch'   }
    {'r_ankle_roll'    }
~~~

Not sure if anyone is planning on working on visualizing this info on the GUI, it could make sense to open an issue on it:  https://github.com/ami-iit/robot-log-visualizer/issues  .

GiulioRomualdi, 2022-04-27:
Hi @gsisinna and @traversaro the name of the joints are already visible on the GUI, however the the matfile @gsisinna is trying to open does not support this new feature. The best way to have the joints name is the visualizer is to migrate the matfile to the new version following what presented here: https://github.com/ami-iit/bipedal-locomotion-framework/tree/master/devices/YarpRobotLoggerDevice#%EF%B8%8F-migrate-mat-files

Uboldi80, 2022-04-27:
Tomorrow, after solving the problem on the strain 2 of the left arm, we will check the temperature of the heatsink to open whether the upgrade is actually served or not.

Uboldi80, 2022-04-28:
this is the situation following the upgrade of the heatsink:
[image]

[image]

As you can see, the temperature difference between 0B3 and 0B4 is about 20 degrees.
I couldn't measure the temperature before the upgrade was fitted, but I guess it could be around 80 celsious.

Uboldi80, 2022-04-28:
I would say that we could still revise the layout for the distribution of the joints on the 2FOCs and also foresee on the new setups a new design of the 2FOC heatsinks which for the future require a higher efficiency in terms of dissipation.

S-Dafarra, 2022-04-28:
I wonder if some optical encoders might be dirty, causing those boards to turn more current into heat :thinking:

maggia80, 2022-04-28:
> I wonder if some optical encoders might be dirty, causing those boards to turn more current into heat 🤔

do you see a big mismatch between the torque and the current in that particular joint?

Uboldi80, 2022-04-29:
Today I took stock with @ emiliano80 to find a solution that allows us to reduce the load on the 2FOCs.
The best thing is to re-propose the E_cub solution in order to align with an official schematic and at the same time verify the new wiring solution.
For this reason, next week I would like to ask for the support of team fix in order to modify the correct files related to the changes described.
@pattacini could I have the support of @davidetome?
I have not yet scheduled the exact day.

Uboldi80, 2022-04-29:
I have scheduled the intervention for Monday 9/05/2022  8:00am to 13:00am
cc. @S-Dafarra @isorrentino @emilianob80 @gsisinna @davidetome 

thanks all

emilianob80, 2022-04-29:
below is the diagram of the new motor connection

[image]
[image]

cc: @S-Dafarra @isorrentino @Uboldi80  @gsisinna @davidetome

Uboldi80, 2022-05-09:
Today with the help of @Gandoo and @davidetome  we have changed the layout of the icub3 torso harness.
now the layout is aligned with that of the Ergocub torso.
following the change of the wiring and the configuration file in hardware / motorcontrol we have running the interface and verified the temperatures.
the situation has visibly improved.
[image]
[image]

closed 2022-05-09 by Uboldi80