robotology/mesh-support #1355: [iCubGenova08] General maintenance

ddetommaso, 2022-04-20:
### Device name 🤖

iCubGenova08

### Request/Failure description

We would need to schedule an intervention for maintenance of the robot in the center in Boggiano Pico, possibly on the 6th or the 20th of May.

### Detailed context

Overall tightening of cables, electrical connections and other generic check up

gsisinna, 2022-04-26:
Hi @ddetommaso, we can schedule the maintenance on the 6th of May. Is it ok for you?

Thanks!

cc @Uboldi80 @Fabrizio69

ddetommaso, 2022-04-26:
Hi @gsisinna 

since a more urgent issue occurred (#1360)  if possible I'd ask you to intervene as soon as possible. 
Any date is fine with us.

Many thanks