robotology/mesh-support #1359: Right arm shoulder joints go in hardware fault when starting the robot

HosameldinMohamed, 2022-04-22:
### Device name 🤖

iCubGenova04

### Request/Failure description

When starting the robot, the left arm goes to hardware fault from the startup. In the logger we see this error.

```
21.597584	ERROR	unable to complete calibration: joint 0 in 'hw_fault status' inside doneRaw() function

```

From `yarpmotorgui` it's possible to reset the fault state of the joints. but when trying to put them in `run` state, all the joints of the robot go in `hardware fault`. Please see the attached video.

### Detailed context

This happened after the intervention in https://github.com/robotology/icub-tech-support/issues/1358# but it seems a different problem.

### Additional context

`yarpmotorgui` state of the joints after trying `run` the `shoulder yaw`

https://user-images.githubusercontent.com/45564317/164756204-75e5aa0f-ed64-4bc5-b941-f5341606b9d2.mp4

HosameldinMohamed, 2022-04-22:
CC @gabrielenava

Uboldi80, 2022-04-26:
We noticed that the 2FOC 0B3 present on the chest, right side presented a sign of breakage on a MOSFET due to the conflict with the eyebolt which had the threaded part too long.
we replaced the board and the problem was solved

closed 2022-04-26 by Uboldi80

emilianob80, 2022-04-26:
@HosameldinMohamed in reference to the comment https://github.com/robotology/icub-tech-support/issues/1359#issuecomment-1109407929

In detail, the damage caused by the pin screw to keep the robot raised. Most likely, during the tests in the wind tunnel, the pin bent and the excessive length caused it to break the mosfet

[image]

cc: @Uboldi80 @gsisinna @Fabrizio69 @maggia80