robotology/mesh-support #1363: The icub robot Yoga suffered a left leg failure during a demonstration

SSSSamZhu, 2022-04-29:
### Device name 🤖

iCubShenzhen01

### Request/Failure description

As the video shows, Icub can't move her left leg.But it's worth noting that this only happened after icub had been in the demo for a while.I can use motorgui directly to control the robot's left leg，each joint can move a little bit.So in the video, ICub appears to have entered a dead zone.Maybe we can recalibrate the encoder

### Detailed context

https://user-images.githubusercontent.com/100264795/165877608-a845d1a7-d452-4a5f-befa-26dfc8fa522a.mp4

SSSSamZhu, 2022-04-29:
hi @Uboldi80 
I rechecked the joints of the robot towards their limits very slowly.And the motorgui readings seem to be fine, changing slowly as the robot's joints move slowly.
Then I noticed something in yoga's .ini initialization file that puzzled me.

1. The parameters of the right leg and the left leg are the same, but the left leg will get stuck when they are demonstrating.
2. The parameters of both joints seem to have an unreachable limit, such as 90 for joint 0 (the maximum seems to reach 82.78 or 83.93) and -0.54 for joint 1 (the minimum seems to reach 0.000), but as mentioned above, only the left leg will stop.

Thank you!

Uboldi80, 2022-04-29:
Ciao @ssssamz
adesso facciamo un altro test:
prima di lanciare la yoga demo apri la motor gui  dopodiché lancia la yoga demo e verifica quale giunto della gamba sinistra va in hfault con finestra di colore rosso o in idle con finestra di colore giallo.
a quel punto fai una foto bella motor gui e postala sulla issue.
dopodiché verifica se é possibile rimettere in RUN il giunto che hai perso durante la demo riportandolo al colore verde.

SSSSamZhu, 2022-05-05:
Hi @Uboldi80
After a few tests, the icub demo seemed to work itself out,icub can work normally now, but we haven't figured out the exact cause yet.We plan to discuss it next time when we meet an issue of the same type.

At present, another difficult problem we are facing is that the solder pad of FPCB of icub's left hand fell off, which was left over when we repaired icub last time.All four of the welded wires in the picture broke and ripped the pads off.I initially think it is difficult to reweld, and the whole FPCB may need to be reprocessed.Or can you provide the corresponding drawings?

[image]

gsisinna, 2022-05-05:
Hi @SSSSamZhu,
before sending other spare parts, we can try to recover the soldering pad with some flux (soldering paste), tin and try to recreate the spot that is now empty to attach the red cable. Do you think you can do this operation?

I look forward to your updates,
Thanks

SSSSamZhu, 2022-05-05:
Hi @gsisinna ,
I tried the welding operation, but because the pad fell off, the solder can't seem to attach to the PCB.By the way,We also have some questions about robot rope installation, such as rope wear and preload.Could we reserve a video call?

gsisinna, 2022-05-05:
We proceed to ask for a quotation for the spare part and as for the videocall we can schedule it for next week. 

We would be available on **Tuesday 10th** or **Thursday 12th**, please let us know if one of those days works for you.

cc @Uboldi80 @Fabrizio69

SSSSamZhu, 2022-05-06:
Hi @gsisinna ,
We are free next **Tuesday** ,We will prepare questions about cable in advance. Looking forward to the videocall on **10th**.

gsisinna, 2022-05-06:
Can you share your institutional mail address? I will send you an invite, thanks!

SSSSamZhu, 2022-05-06:
Hi，
OK，This is my mentor's email address henricli@tencent.com，look forward to our meeting on Tuesday！

gsisinna, 2022-05-10:
Hi @SSSSamZhu,
in order:

- 1) for everything related to purchase proposals and spare parts you must refer to [Andrea Derito](https://www.iit.it/it/people-details/-/people/andrea-derito) by sending an email with cc [Marco Maggiali](https://www.iit.it/it/people-details/-/people/marco-maggiali). I can give you the IIT internal code for the particular part or board if you need it.

- 2.1) Regarding tendon preloading, no tools or procedures are currently available for cable pre-tensioning, but reference is always made to the manufacturer's construction specifications for maximum allowable stresses, and in the design phase this was considered for pulley sizing and various joint loads. If necessary I can refer to the team of mechanics who will certainly be able to give a more complete answer and provide more technical details.

- 2.2) For the maximum allowed speed I have consulted with colleagues, and there is a reference that I think is useful to answer this question and that indicates upper limits for the speed of joints ([HERE](https://github.com/robotology/community/discussions/194)). 

> In this respect, I would say that we shouldn't overcome **200 deg/s peak** and **50 deg/s mean velocity**. The peak I'm referring to, for example, could be the maximum velocity delivered in a **minimum-jerk position profile**.

I suggest that you take these limits with caution and place yourself well below these values and then try to increase the speed to evaluate the response of the system. I also suggest to always rely on the control modes indicated in the official documentation and follow the guidelines to avoid failures and breakage of critical components.

Low level PIDs are linear and therefore not equipped with dedicated vibration components, so your case should be studied ad hoc to understand if and in which case vibrations are present.

- 2.3) I can share with you a technical dossier about it for accuracy and repeatability analysis. 

And finally for the last question, clearly iCub is equipped with encoders that it uses for closed loop control and there are tests to evaluate the consistency of the encoders and any drift before the release of a new robot. Everything is then tested together with the tendons cables through functional tests, and if you need particular performances we remain available to evaluate upgrades or eventual modifications. For the control bandwidth or frequency analysis I can later share a second technical report about it and extrapolate the necessary information.

Thanks,
Gabriele

cc @Uboldi80 @Fabrizio69 @maggia80

gsisinna, 2022-05-10:
Internal IIT code for left palm skin (as mentioned during the call):

**4702 . B - MMSP_L - Electronic board prototype, multimodal skin for the palm (LEFT)**

cc @Fabrizio69

SSSSamZhu, 2022-05-20:
Hi，
We have counted our spare parts inventory, and we need 1815 and 2390 cables.You can quote together with the spare parts mentioned above.The contact address is still the same as before: email address henricli@tencent.com
Thanks！
cc @Fabrizio69@maggia80

Uboldi80, 2022-05-20:
hy @SSSSamZhu 
How much cable will you need? 1m? 2m per type?
3m per type?
then 2 types of cable and the left palm (skin)
Correct?

SSSSamZhu, 2022-05-23:
Hi,We need 3m per type of cable.We confirm that we only need these two types of cable and the left palm.

Uboldi80, 2022-05-23:
Ok, but i think along with the 2390 cable you should also take some crimp to create eventually some tendon.

Uboldi80, 2022-05-23:
Hy @andreaderito
we need a quotation for the spare parts which I will describe below:
3m of 2390 cable
3m of 2391 cable
3m of 1815 cable
10 crimp 3492
1 skin left palm 4702.B

cc. @Fabrizio69

closed 2022-08-10 by SSSSamZhu