robotology/mesh-support #1364: Trembling left hand

Matilde-Antonj, 2022-04-29:
### Device name 🤖

iCubGenova03

### Request/Failure description

If you use the CartesianSolver to specify the target position, the left hand manages to achieve the specified position, but it trembles.

### Detailed context

Test A (Left arm):
- the target position was specified through the Cartesian Solver
- the left hand achieved the target position, but it continues trembling
- if the trembling left hand is touched, it is possible to feel a vibration

Test B (Right arm):
- the specular target position was specified through the Cartesian Solver
- the right arm achieved the target position without trembling

### Additional context

https://user-images.githubusercontent.com/103580025/165988719-dc84283f-06db-4d48-b1c0-c42ae7e05a52.MOV

Uboldi80, 2022-05-04:
Hy @Matilde-Antonj 
We are trying to understand what the behavior of the wrist is due to.
probably come by to perform an operation on the robot next week Tuesday or Thursday.
we will update you on the exact day.

Matilde-Antonj, 2022-05-05:
Perfect, thank you!

Uboldi80, 2022-05-10:
Hy @Matilde-Antonj 
Me and @gsisinna  come to Erzelli this Thursday 12/05.
for you are ok?

Matilde-Antonj, 2022-05-10:
It's perfect for me, see you on Thursday!

closed 2022-05-12 by gsisinna