robotology/mesh-support #1367: Migrating to Robotology-Superbuild

b-w-wilson, 2022-05-09:
### Device name 🤖

iCubEdinburgh01

### Request/Failure description

We wish to update the whole iCub environment software to the latest version applicable for our robot.

### Detailed context

We are working on a research project that will involve collaboration with several other site's iCub's, and we wish to update all the robots to the latest version to ensure similarity across development. 

The iCubEdinburgh01 robot seems to be an older version, but it seems to be running the icub OS rather than LIVE version. (I don't know how to properly confirm this, nor do I know what hardware version the robot is unfortunately!)
It however doesnt seem to be running the robotology-superbuild, and we would rather avoid updating all the packages manually.

I have only just began working with the robot here in Edinburgh, however I have had warnings from the prior users, that the Edinburgh robot should never have its software updated for some reason, relating to someone previously attempting to update it and having issues, leading to some configuration files being modified.

So for this warning, plus the note on the wiki to coordinate with icub-tech-support when migrating to the superbuild, I would greatly appreciate any assistance with updating the robot to the latest software version!

gsisinna, 2022-05-11:
Hi @b-w-wilson ,
thank you for writing to our support. I would say to start by gathering some additional info about your robot and come up with a solution customized to your needs taking into consideration the warnings of your colleagues.

For information regarding the serials and version of your robot you can refer to the **RobotBazaar** website, where you can search for your robot: [HERE](https://robot-bazaar.iit.it/robots/iCubEdinburgh01).

Instead, [here](https://icub-tech-iit.github.io/documentation/sw_installation/icub_head_superbuild/#superbuild-installation-on-icub-head) you will find all the necessary instructions for installing the robotology-superbuild, step by step.

There are several things to pay attention to, and among them in particular:

1. The **.bashrc_iCub** will need to be updated with the new one related to robotology_superbuild, and not the old one where all modules were compiled and built separately. Otherwise all environment variables will be wrong and you will not be able to start contexts and applications.

2. I don't know if you have made any changes to the robot configuration with respect to the official robots-configuration [repo](https://github.com/robotology/robots-configuration/tree/master/iCubEdinburgh01), so I recommend that you save or send us your calibration files so that you can keep the system stable after upgrading to the robotology-superbuild.

It would be helpful to have additional details, such as those you can see as soon as you open a shell on icub-head such as the full version of the operating system and the folder structure in **$ROBOT_CODE**.

If the official documentation is not enough, we can also arrange a videocall to provide remote assistance and evaluate together what to do.

Thanks!

cc @Uboldi80 @Fabrizio69

b-w-wilson, 2022-05-16:
Hi @gsisinna,

Many thanks for your reply. Hope you're well, its been a little while!

For the details needed first:

Full robot OS version: iCub OS version 7.0-20160902 (based on Ubuntu 16.04.1 LTS (GNU/Linux 4.4.0-47-lowlatency x86_64)
(I included the linux version as it includes the processor architecture)

Inside $ROBOT_CODE:
- iCubContrib
- icub-contrib-common
- icub-firmware-build
- icub-firmware-shared
- icub-main
- robots-configuration
- yarp

If the robots-configuration is the repo that contains all configuration files, we have apparently made some small changes to the right_arm-calib.xml, and added 2 other ini files for either camera, however apparently this repo is also on quite an old commit (2018).

Also a quick question, can we update the icubserv laptop with robotology-superbuild too? As I assume the software on there is running many versions behind too.

Many thanks!

gsisinna, 2022-05-17:
Hi @b-w-wilson, 
thanks for your feedback! 

> Also a quick question, can we update the icubserv laptop with robotology-superbuild too? As I assume the software on there is running many versions behind too.

The best thing would be to align all the software, both for the head and the laptop, and use the robotology-superbuild. 

To install the robotology-superbuild on your **laptop** you can fin a complete guide following this steps:

1. [Generic iCub environment machines](https://icub-tech-iit.github.io/documentation/icub_operating_systems/other-machines/generic-machine/)
2. [The iCub Console Server](https://icub-tech-iit.github.io/documentation/icub_operating_systems/other-machines/icub-server-laptop/)
3. [Robotology-superbuild](https://github.com/robotology/robotology-superbuild#superbuild)

**Always consider your version of Ubuntu installed on the laptop.**

> If the robots-configuration is the repo that contains all configuration files, we have apparently made some small changes to the right_arm-calib.xml, and added 2 other ini files for either camera, however apparently this repo is also on quite an old commit (2018).

I would still recommend saving the configuration folder locally or sending the entire folder to us to update upstream in robots-configuration.

For **installing the robotology-superbuild on the icub-head** I will refer to the same [comment](https://github.com/robotology/icub-tech-support/issues/1367#issuecomment-1123568101) I sent you before.

Thanks!

closed 2022-07-27 by Uboldi80