robotology/mesh-support #1371: VelocityControl mode not working on iCub, but working on iCubSim

lunagava, 2022-05-13:
### Device name 🤖

iCubGenova02

### Request/Failure description

We are trying to control in velocity mode but it is not working with the iCub, whereas the same code works in simulation. We are asking for someone next week (possibly ASAP) to check together with us the reason why it is happening.

### Detailed context

We are wondering if some configuration files changed due to recent updates.

pattacini, 2022-05-13:
Hi @lunagava 

Triaging issues, which includes assigning tasks to developers, should be left to iCub Tech maintainers 😉

Regarding this request, my hunch is that it's unlikely that it may be impacted by changes in the configuration files.

It would be a good starting point to share your code via a Gist as per the [guidelines](https://github.com/robotology/community/blob/master/.github/SUPPORT.md). Actually, the [community discussions](https://github.com/robotology/community/discussions) could be a better place for this request.

Also, I'll be on vacation the next week.

lunagava, 2022-05-13:
Sorry @pattacini, I was not sure where to put the issue (if icub-tech-support or community discussion) and then I made the wrong choice. I'll close the issue here, and redo a new post inside community.

closed 2022-05-13 by lunagava