robotology/mesh-support #1372: General Maintenance of R1 SN003

randaz81, 2022-05-16:
### Device name 🤖

R1SN003

### Request/Failure description

Important:
- check the status of the elbow belt in right arm (repaired during an emergency @fbiggi, but the belt was not tightened)
- check the status of the left wrist (broken)

Optional:
- check the status of the neck belt (not tightened)
- check the status of both the hands (not closing properly, tendon broken or recalibration needed)the 

### Detailed context

Required before 22/05. The robot will be packed for shipment on the 23rd evening.

maggia80, 2022-05-16:
issue occurred in the right upperarm: [image]
[image]

parts delivered for replacing the broken shaft:

[image]

fbiggi, 2022-05-18:
inizio intervento ore : 0600 
termine intervento ore: 13:30 
@Uboldi80  se abbiamo ok da @randaz81 chiudiamo le issues

c @Fabrizio69 @maggia80

Uboldi80, 2022-05-18:
I just speeched with @marco93 and then, seems the robot work well.

closed 2022-05-18 by Uboldi80