robotology/mesh-support #1375: iCub right arm joint 6

pfdominey, 2022-05-19:
### Device name 🤖

iCub Lyon ->  Dijon 

### Request/Failure description

Joint 6 right arm has lost control.  The wrist is slack when the motors are on.
[Uploading yarprunlog_19_05_2022_11_30_45.log…]()

### Detailed context

joint 6 in the write arm cannot move

examining the wrist, there is slack in this yaw movement
it looks like there might be a cable that has fallen off its round holder

### Additional context

[Uploading yarprunlog_19_05_2022_11_35_31.log…]()

### How does it affect you?

we use ctpServices to control the robot.  When we execute our sequences, the wrist behaves in a random fashion.

Uboldi80, 2022-05-19:
Hy @pfdominey 

Can you send us some photo of the wrist (from different point of view)?
thanks

pfdominey, 2022-05-19:
here is the link to a few more photos

https://photos.app.goo.gl/evb5N9MueERLXw3k9

Uboldi80, 2022-05-19:
Good morning @pfdominey 

At the following link you can find the instruction to fix the wrist cable:
https://icub-tech-iit.github.io/documentation/icub_tendons/wrist_hand_cabling_V1/#112-cabling

Uboldi80, 2022-05-20:
Hy @pfdominey 
The cable I indicated to you is the one that most likely came out of the office.
to make sure of this, take the cable and check that the crimp is present on the end of it, which should be repositioned in its seat.
[image]

Uboldi80, 2022-05-20:
After doing this check let me know if the tendon is broken or not.

pfdominey, 2022-06-01:
When we pull gently on the cable it does not come out, suggesting that it is not broken.
However it is very difficult to access.

If we unscrew the two screws that hold the hand onto the wrist, will this give us more access?

pfdominey, 2022-06-01:
[image]

It looks like the cable is well connected at both extremities, but the pully is poorly positioned.

Do you think that we can turn on the robot, and try to use the motor gui to re-position the pulley?

pfdominey, 2022-06-01:
[image]

Here is another photo showing the slack in the cables.  Is there a way to correct this by turning the motors with the motor gui?

pfdominey, 2022-06-01:
We have the impression that the cable fell off the pully, and then the motors got put into an unnatural position.  Ideally we can just reseat the cable on the pulley and the motors will pull everything back into place?

gsisinna, 2022-06-01:
Hi @pfdominey, please refer to [this](https://icub-tech-iit.github.io/documentation/icub_tendons/wrist_hand_cabling_V2/#13-wrist-passive-tendons) section for wrist cabling and repair.

As you can see in the next pictures, you need to play with the tensioner in order to properly fix this passive tendon:
[image]
[image]

pfdominey, 2022-06-02:
Thanks for this.  However, we are really neophytes at working on this.  Could you please write down a step by step procedure, or we could have a visio conference?
Thanks

gsisinna, 2022-06-07:
Hi @pfdominey, we will send you a video for your hardware version and a step by step procedure.

We already have your mail address.

Thanks,
Gabriele

cc @Uboldi80

pfdominey, 2022-06-07:
Thanks!  Looking forward to seeing the video!
cheers
Peter

gsisinna, 2022-06-07:
I updated also the reference for your wrist version in the previous comment (https://github.com/robotology/icub-tech-support/issues/1375#issuecomment-1143731749). The step by step procedure we use is [this](https://icub-tech-iit.github.io/documentation/icub_tendons/wrist_hand_cabling_V1/#122-cabling) one.

pfdominey, 2022-06-07:
Before fixing the wrist is it ok for us to turn on the robot and do some work with the left arm?

pfdominey, 2022-06-08:
Thanks for your info.
When can we expect the video?
We are waiting to run experiments :)
thanks
cheers
Peter

Uboldi80, 2022-06-08:
Good morning @pfdominey 
Below are the 4 videos that should help you solve the problem.
In this case I assumed that the tendon has only come out of its seat and therefore has not broken.
if it is torn then it will need to be replaced.
let me know.

https://user-images.githubusercontent.com/80786976/172585570-8577ce07-073d-42f3-92ad-334a2c1fdc0f.mp4

https://user-images.githubusercontent.com/80786976/172585341-9509124a-276a-4b52-8f9a-9123962ceebb.mp4

Uboldi80, 2022-06-08:
https://user-images.githubusercontent.com/80786976/172586548-36cecfde-da79-4e0e-bccd-7728b3f50db9.mp4

Uboldi80, 2022-06-09:
yes, follow the instructions I sent you and check that the tendon is not torn.

pfdominey, 2022-06-09:
the cable is not broken.
we are able to tighten the tensioner and get things correct
if we then move the hand passively this seems to reintroduce a slack in the cable.
We are tempted to repositon the cable, and then turn on the robot.
do you agree?

pfdominey, 2022-06-09:
Also, is there a coupling between cable 9 and 10?  should we also tighten cable 10?

Uboldi80, 2022-06-09:
Hy @pfdominey 
before to restart the robot thight the wrist cable through the screw of the tensioner.
let me know

pfdominey, 2022-06-09:
we now have the cables tensioned

pfdominey, 2022-06-09:
should we try to start the robot?

pfdominey, 2022-06-09:
before we start the robot, the cable keeps falling off the pulley, because of our own passive movment of the wrist

pfdominey, 2022-06-09:
once again, when we turn on the robot, the cable becomes loose again.
[image]

Question:

Because the cable was not broken, we did not remove the screw and the plate that holds the cable, as you illustrated in Video 3.

Do you think this is necessary?

Uboldi80, 2022-06-09:
no, it is not necessary, you can also try without removing the screw and the plate

pfdominey, 2022-06-09:
We now have a broken cable.  See attached photo.
[image]

What is the procedure to ask to  a technician to come here to Dijon to fix the robot?  We can pay travel expenses and repair expenses if necessary.  Our problem is urgent.  Normally we should be running a major experimental campaign now.
Also, the robot was only recently repaired and sent from IIT.
Please let us know.
cheers
Peter

gsisinna, 2022-06-10:
Hi @pfdominey,
a possible temporary solution might be to exclude the wrist from the calibration of the right arm and immobilize it mechanically so that you can continue with your experiments.

To do this you need to go edit the robot configuration files on the head, then connect in ssh and navigate from terminal to the desired file.

The configuration files are present in several copies, but to make the changes permanent it is necessary to edit the source files and afterwards install those files with simple commands.

### Goal: Edit [this](https://github.com/robotology/robots-configuration/blob/66cc6d24cbde1f825f3e202248698f7417a3f7b9/iCubDijon01/calibrators/right_arm_calib.xml#L28) line in the calibration file.

1. From the robot laptop: `ssh -X icub-head`
2. Once you're inside the icub-head: `cd /usr/local/src/robot/robotology-superbuild/src/robots-configuration/iCubDijon01/calibrators`
3. Here you can use your favourite text editor to edit the `right_arm_calib.xml` file, for example: `nano right_arm_calib.xml`
4. Modify line [28](https://github.com/robotology/robots-configuration/blob/66cc6d24cbde1f825f3e202248698f7417a3f7b9/iCubDijon01/calibrators/right_arm_calib.xml#L28) into this: `       <param name="CALIB_ORDER">(0 1 2 3) (4 7)</param>` (I removed joint 5 and 6 from the calibration)
5. Save and close the file (CTRL+ X with nano editor, and then Y + ENTER)
6. Type: `cd /usr/local/src/robot/robotology-superbuild/build/src/robots-configuration`
7. Install the files from the source: `make install`

Now, when you launch the `yarprobotinterface` from the `icub-head` cluster you should use the new configuration files and the wrist will be excluded from the calibration.

If something doesn't go well, we can set up a call and try writing these same commands.

Thanks!

cc @pattacini @Uboldi80 @Fabrizio69

pfdominey, 2022-06-10:
Thanks
Three questions
1.  could you check that ctpServices for the right arm will work after we exclude the wrist from calibration?
2. is it ok to physically make a kind of splint/bandage around the wrist so that the wrist does not flop around / move when we are doing pointing movments?
3. what would happen if we do not exclude joints 5 and 6 from calibration?  Does that cause any harm to the robot?
thanks

pattacini, 2022-06-10:
We exchanged info through email ✉️ too, so let me report here for the record that:

>1. could you check that ctpServices for the right arm will work after we exclude the wrist from calibration?

I think that `ctpService` is robust enough to control the arm with the uncalibrated wrist.
 
This is because, from this [relevant control snippet](https://github.com/robotology/icub-main/blob/master/src/modules/ctpService/main.cpp#L269-L284), ctpService won’t take any resolutive action when it’ll read that the joint is not in the correct mode, aside from printing out a simple error message.

ctpService will try to command it with the `positionMove` service but the motor boards will simply skip it.

>2. is it ok to physically make a kind of splint/bandage around the wrist so that the wrist does not flop around / move when we are doing pointing movements?

Fine 👍🏻 

>3. what would happen if we do not exclude joints 5 and 6 from calibration? Does that cause any harm to the robot?

Trying to calibrate broken coupled joints is risky as the broken cable will be causing damage.

pfdominey, 2022-06-21:
Hi
in the motor gui joints 5 and 6 in the right arm appear normal, green

pfdominey, 2022-06-21:
I verified that I properly modified the line in the file

<param name="CALIB_ORDER">(0 1 2 3) (4 7) </param>

pfdominey, 2022-06-21:
[image]

gsisinna, 2022-06-21:
Thanks for the feedback @pfdominey. Can you tell us the full path of the file you modified? (On the icub-head)

pfdominey, 2022-06-21:
as instructed:

cd /usr/local/src/robot/robotology-superbuild/src/robots-configuration/iCubDijon01/calibrators

pfdominey, 2022-06-21:
Things seem to be working.  joints 5 and 6 are keeping stable at -16 and -10 respectivley when the robot is pointing.

gsisinna, 2022-06-21:
> as instructed:
> 
> cd /usr/local/src/robot/robotology-superbuild/src/robots-configuration/iCubDijon01/calibrators

Did you install this files with:

1.  `cd /usr/local/src/robot/robotology-superbuild/build/src/robots-configuration/iCubDijon01`
2.  `make install`

pfdominey, 2022-06-28:
I followed the instructions:

https://github.com/robotology/icub-tech-support/issues/1375#issuecomment-1152127176

6. Type: cd /usr/local/src/robot/robotology-superbuild/build/src/robots-configuration
7. Install the files from the source: make install

so the make install command was executed in the robots-configuration directory and not the iCubDijon01 directory

gsisinna, 2022-06-28:
Hi @pfdominey, actually you can launch the `make install` command to install all the configurations files (from the `robots-configuration` directory) or only for Dijon01 (from `/robots-configuration/iCubDijon01` folder).

Now you should have your calibrators installed for the `yarprobotinterface`

pfdominey, 2022-06-28:
what is strange now, is that when I execute

cd /usr/local/src/robot/robotology-superbuild/src/robots-configuration/iCubDijon01/calibrators

I get an error!!

icub@pc104:~$ cd /usr/local/src/robot/robotology-superbuild/src/robots-configuration/iCubDijon01/calibrators
-bash: cd: /usr/local/src/robot/robotology-superbuild/src/robots-configuration/iCubDijon01/calibrators: No such file or directory
icub@pc104:~$ 

same for 

cd /usr/local/src/robot/robotology-superbuild/build/src/robots-configuration/iCubDijon01

icub@pc104:~$ cd /usr/local/src/robot/robotology-superbuild/build/src/robots-configuration/iCubDijon01
-bash: cd: /usr/local/src/robot/robotology-superbuild/build/src/robots-configuration/iCubDijon01: No such file or directory
icub@pc104:~$

gsisinna, 2022-06-28:
iCubDijon01 mounts a pc104 that remotely mounts the filesystem, so it seems that the icub-head boot and mount procedure failed or is stuck.
How was the robot connected with the laptop and power rack? I mean for the ethernet part especially.

pfdominey, 2022-06-28:
OK I can dir into those directories now.
Sometimes I have to reboot the pc104.

when you say  "iCubDijon01 mounts a pc104 that remotely mounts the filesystem" where is iCubDijon01 running?

thanks!

gsisinna, 2022-06-28:
Ok, it is necessary to wait a few minutes at startup for models with pc104 for the filesystem to be mounted remotely.

Here is the [documentation](https://icub-tech-iit.github.io/documentation/icub_operating_systems/pc104/the-icub-live-details/#nfs-configuration) for the Network File System (NFS) used.

And here the [reference](https://icub-tech-iit.github.io/documentation/icub_operating_systems/pc104/the-icub-live-details/#the-mount-script) to the mount script to the icubsrv (10.0.0.1)

gsisinna, 2022-06-28:
For all requests regarding on-site interventions please refer to [Marco Maggiali](https://www.iit.it/web/guest/people-details/-/people/marco-maggiali), thank you.

andreaderito, 2022-08-03:
Dear @pfdominey as discussed via email, we'll ship the robot here at IIT-CRIS headquarters for the intended repair. Expected arrival at CRIS around mid August 2022.
cc @gsisinna @Uboldi80 @Fabrizio69 @maggia80

GiacomoVassallo13, 2022-08-23:
Good morning, the tendon of the wrist of the right hand has been restored. For now, I do not find there are other mechanical interventions to be carried out

cc:  @Fabrizio69 @maggia80

pfdominey, 2022-08-27:
there is a sh script in /home/pointing_sequences protocol_balanced_fixing.sh
seqA, seqB etc.
These require the ctpservice is activated for left_arm, right_arm, head and torso
If you want to try testing these sequences
It was using these sequences that the wrist was broken
thanks!

gsisinna, 2022-09-06:
Hi @pfdominey,
we reviewed the entire wrist subassembly and replaced the cable involved for the movement of joint 6. 
The robot was also tested with the sequences you mentioned with the ctpService, and I saw that in particular seqA has critical positions for any iCub wrist, so these may have overstressed the V1 hands and pulleys involved in the movement.

Here some videos:

https://user-images.githubusercontent.com/36999962/188637652-20a5de90-0cce-4f4b-8ed5-ecd725e9416d.mp4

https://user-images.githubusercontent.com/36999962/188639038-749631fe-4ae6-47bc-9908-38e155320477.mp4

I'll leave the rest to @andreaderito for shipping and closing this issue.

cc @maggia80 @Fabrizio69

pfdominey, 2022-09-07:
Hi @gsisinna 
Thanks for taking care of our iCub!
In the video I noticed that the head is not coordinated with the pointing - I assume you did not turn on ctpservice for the head?
Also, for the bad movements in SeqA, can you suggest a method that I can use for systematically eliminating such joint angles?
You may have noticed that each sequence is made up of several idndividual pointing movements, so i need to identify the movements that have bad joint angles.
Were you able to detect whcih joint in the wrist was being stressed?  and what i the safe range of angles?
thanks for your help.  We dont want to break it again!
cheers
Peter

pfdominey, 2022-09-21:
Hi @gsisinna 
Thanks for taking care of the iCub! 
 In the .sh file (/home/pointing_sequences protocol_balanced_fixing.sh seqA, seqB ) there are 12 pointing movoements  (point1Right - point12Right) that are distributed in the different sequences SeqA SeqB etc.

It would be great if you could "As for the joint angles I will try to compare those you sent to the robot with the script and those present as limits in the low-level configurator"

The iCub is back in Dijon and we will be getting him set up and starting experiements again soon.

If it is too complicated to look at all the movements, if you could just give me the limits then I can check and correct.

thanks
cheers
Peter

gsisinna, 2022-09-21:
Hi @pfdominey,
[here](https://github.com/robotology/robots-configuration/blob/fec822c2d139b693390e2a89133e3f0b45856cb1/iCubDijon01/hardware/motorControl/icub_left_arm.xml#L56) you can find all the low level limits for joint angles. 

For example, for the wrist:

```
<param name="AxisName"> "l_shoulder_pitch" "l_shoulder_roll" "l_shoulder_yaw" "l_elbow" "l_wrist_prosup" "**l_wrist_pitch**" "**l_wrist_yaw**" "l_hand_finger" </param>

<param name="jntPosMin">          -95.5         0             -30           15            -90           **-80**           **-20**           0             </param>       
<param name="jntPosMax">          10            160.8         75            106           90            **25**            **25**            60            </param>

```

Take this values as reference, thanks!

gsisinna, 2022-09-21:
I am closing this issue as the goods have been shipped and have been received at the facility.

closed 2022-09-21 by gsisinna

pfdominey, 2022-09-22:
We are getting ready to set yup the iCub.  I cannot find in the shipping box the four small screws that hold the robot.
If the pin is in place is that sufficient?
[image]

pfdominey, 2022-09-22:
[image]
closer view

gsisinna, 2022-09-22:
Hi @pfdominey, 
it is our understanding that the screws are in the box and were included, they should be in a small bag attached to the packing material.

If they are not present these are common screws that you can buy at any hardware store:

4 screws, 3x12 cylindrical hollow head hexagonal

pfdominey, 2022-09-22:
yes - they were there!  I have now installed them.  I must go now. Tomorrow morning I will turn on the robot.  I remember the procedure from the last time, so unless you suggest otherwise, I can do the startup without your assistance.
thanks again