robotology/mesh-support #1377: Right ankle joints go sometimes in hardware fault

gabrielenava, 2022-05-27:
### Device name 🤖

iCubGenova04

### Request/Failure description

While the robot is balancing on two feet the right ankle joints sometimes go in hardware fault. There are two errors messages related to right ankle board in the yarplogger:

1) strain timeout
2) AEA encoder invalid data

```
 38.584629 <ERROR>  from BOARD 10.0.1.9 (right_leg-eb9-j4_5), src LOCAL, adr 0, time 8953s 231m 25u: (code 0x04000003, par16 0x0000 par64 0x0000000000000000) -> DEBUG: tag03 + strain timeout
 39.403638 <ERROR>  from BOARD 10.0.1.9 (right_leg-eb9-j4_5), src LOCAL, adr 0, time 8954s 58m 47u: (code 0x02000008, par16 0x0001 par64 0x000000000000000b) -> MC: AEA encoder invalid data. Hardware problem in the magnetic position sensor of the joint caused invalid position readings. par16 = AEA port (msb) and ID of joint (lsb). + .
```

### Detailed context

here the full logger:

[log_icub-head_yarprobotinterface_45612.txt](https://github.com/robotology/icub-tech-support/files/8785446/log_icub-head_yarprobotinterface_45612.txt)

### How does it affect you?

It affects the experiments programmed for the iRonCub rush.

cc @DanielePucci

Uboldi80, 2022-05-30:
I found several connectors with problems:
the E11 cable on the EB11 had a problem on the shield.
The 9N1 cable on the 3B10 can2 had a nearly torn cable
The 9N2 strain cable on the EB9 had the red cable nearly torn.
I restored the wiring and turned on the setup to test it.

closed 2022-05-30 by Uboldi80

HosameldinMohamed, 2022-05-31:
> Hi @Uboldi80, after your intervention the issue happened again when the turbines were ON. Now the issue is more frequent when there's relatively high temperature.
> 
> We also noted that the 2 errors mentioned [above](https://github.com/robotology/icub-tech-support/issues/1377#issue-1250530312) (Strain timeout and hardware fault) happen at the same time.
> 
> Here are the logs from yesterday and today's experiments (When the turbines where ON):
> 
> [yarprunlog_30_05_2022_19_37_24.log](https://github.com/robotology/icub-tech-support/files/8807364/yarprunlog_30_05_2022_19_37_24.log)
> 
> [yarprunlog_31_05_2022_18_53_24.log](https://github.com/robotology/icub-tech-support/files/8807362/yarprunlog_31_05_2022_18_53_24.log)

Hi @Uboldi80, I wrote the above comment with iRonCub account. Sorry it was on the fly.