robotology/mesh-support #1380: Left index finger does not calibrate

S-Dafarra, 2022-06-09:
### Device name 🤖

iCubGenova09

### Request/Failure description

The left index finger does not calibrate

### Detailed context

When running the ``yarprobotinterface``, the left index finger does not calibrate.

Uboldi80, 2022-06-10:
Today i do the calibration of the joint 12.
now seems to work well.

S-Dafarra, 2022-06-10:
Unfortunately, even if the calibration worked well, when we tried to move it straight from motorgui went into hardware fault (hardware limit reached). Then, after restarting the robot, the finger had again calibration issues.

Uboldi80, 2022-06-13:
this morning I reviewed the calibration of the left distal index, also I reviewed the calibration of the middle distal right hand and pinkie right hand, as sometimes they did not calibrate correctly.