robotology/mesh-support #1382: Screw blocked inside left ear

lunagava, 2022-06-20:
### Device name 🤖

iCubGenova02

### Request/Failure description

We disassembled the helmet for a photoshoot with the European commission. The screw was longer to work better with the iCub helmet and a portion remained outside making it unaesthetic. 

### Detailed context

We tried to unscrew, but we need further assistance to solve it.

GiacomoVassallo13, 2022-06-20:
Buogiorno, la vite bloccata sulla testa di iCub `e stata rimossa.  Si tratta delle viti che tengono la faccia, collocate dalle orecchie. 

[image]

Queste erano state sostituite con viti a testa cilindrica piu lunghe per sorreggere il supporto camera. Cio ha fatto si che una volta rimosso il cascetto, nel tentativo di stringere la vite quest ultima si `e bloccata facendo torcere a sua volta il supportino in acciaio, cod. 5203 (revisione zero) Alias RC_IIT_023_P_018

[image]

Quest' ultimo avvitato al telaio tramite il pezzo in alluminio Alias: RC_IIT_023_P_012 cod. 5198 ( revisione non identificata ma obsoleta ) con filettatura M3.

[image]

Questo ha causato la rottura della filettatura interna al pezzo di alluminio, in questo modo vite e supporto giravano  a vuoto.

[image]

cc: @lunagava @Fabrizio69

GiacomoVassallo13, 2022-06-20:
Per risolvere il problema sono stato costretto a tagliare al testa alla vite, cosi da poterla spingere all interno della testa cosi da poter aprire la cover ed strarre il supporto in acciaio.
A quest ultimo sono state ripassare le filettature e attualmene risulta in buono stato.

[image]
[image]

cc: @Fabrizio69

GiacomoVassallo13, 2022-06-20:
Il pezzo in alluminio Alias: RC_IIT_023_P_012 cod. 5198 `e stato ripristinato, il foro e' stato allargato ed e' stato inserito un helicoil.
I  pezzi sono stati rimontati sul telaio. Duratente l' operazione il connettore delle espressioni facciali e' stato rifatto poiche rimasto danneggito. Attualmete `e stayo tutto ripristinato e le cover faccia rimontate sul robot.
L intervento `e concluso procedo con la chiusura della issue.

cc: @lunagava @Fabrizio69

closed 2022-06-20 by GiacomoVassallo13