robotology/mesh-support #1383: Right ankle joints go sometimes in hardware fault (part 2)

gabrielenava, 2022-06-21:
### Device name 🤖

iCubGenova04

### Request/Failure description

follow up of https://github.com/robotology/icub-tech-support/issues/1377

the problem is not resolved, and the robot ankle keeps going in Hardware Fault. We suspect it is related to an increase of temperature. 

### Detailed context

@HosameldinMohamed  and @vpunithreddy investigated the problem, can give more details on it.

### How does it affect you?

it happens during iRonCub experiments, it is difficult to perform take off experiments  with the ankle in fault.

HosameldinMohamed, 2022-06-21:
> @HosameldinMohamed and @vpunithreddy investigated the problem, can give more details on it.

After the intervention in https://github.com/robotology/icub-tech-support/issues/1377 on `30/May` and another one on `09/June`. The problem still occurs.

However, after `09/June` intervention, where the right-foot FT was replaced, the issue happens less frequently than before.

We did a test yesterday, where we placed the robot outdoor and the leg covers were on (both iCub covers and iRonCub heat-resisting covers).

After a long time (more than an hour) the issue occurred (hardware fault on right ankle joints + right foot FT freezes at the same time).

We removed iRonCub heat-resisting cover and left the joints in `idle` for some time, and the issue disappeared. We put the joints in `run` again, and the issue didn't appear for some time.

After that, we placed the robot on the ground, while the ankle pitch was slightly bent to apply a higher load on the motor. In this case, after a short period, the issue happened again.

In summary, we managed to trigger the issue in 2 ways, having the legs covered with both covers, and having the ankle motors under load.

@vpunithreddy feel free to modify the comment!

gsisinna, 2022-06-28:
Thank you guys for the valuable feedback. If it is okay with you I would continue to monitor between today and tomorrow during your experiments and if necessary we are available to make an intervention assessment based on your time schedule and needs.

cc @Uboldi80

gsisinna, 2022-06-28:
Today I spent the afternoon debugging the setup and I replaced also a 2FOC board:
[image]
[image]

I noticed high temperatures on the 2FOC boards controlling the ankle joints (after some minutes in standing position or inclined). I found also different values for the PID parameters of the right leg ([HERE](https://github.com/robotology/robots-configuration/blob/18d40d12fb86dfdc123721523c13fcec048665b1/iRonCub-Mk1/hardware/motorControl/right_leg-eb9-j4_5-mc.xml#L50)) and for the left one ([HERE](https://github.com/robotology/robots-configuration/blob/18d40d12fb86dfdc123721523c13fcec048665b1/iRonCub-Mk1/hardware/motorControl/left_leg-eb7-j4_5-mc.xml#L50)).

I checked all the connectors and the crimps:
[image]

The problem is still there, and we need to understand how and whether high temperature is involved in the chain of errors that come up.

cc @Uboldi80

gsisinna, 2022-06-29:
A possible attempt could be to add an additional heat sink in the two 2FOCs that control the joints of both ankle pitches.
This could be a problem of the boards being undersized for the current drawn and to maintain the weight of the turbines, which shift the robot's center of mass.

HosameldinMohamed, 2022-06-29:
Thanks for the thorough investigation @gsisinna!

I am curious to know which values we are seeing in `yarpmotorgui` because I thought they are the real encoder values. And they seemed logical.

gsisinna, 2022-07-05:
Today with @Uboldi80 we fixed the L-core readings for the right ankle pitch joint encoder. We tested also the setup in a stressful position for the ankle (1 hour) and we didn't have hardware fault or strain timeouts.  We will wait for other tests from the guys (for example the demo yoga).

Some screenshot for the mean current values and temperature readings:
[image]
[image]

cc @gabrielenava @HosameldinMohamed @vpunithreddy

DanielePucci, 2022-07-08:
@gsisinna @Uboldi80 yesterday I had the chance to perform some experiments with the team and the problem keeps occurring unfortunately.

gsisinna, 2022-07-13:
As reference, [here](https://github.com/robotology/icub-tech-support/issues/1394#issuecomment-1181560473) other fixes regarding this issue.

cc @Uboldi80

gsisinna, 2022-07-27:
@gabrielenava any feedbacks?

gabrielenava, 2022-07-27:
yes @gsisinna, yesterday we tested the robot while hanged on the crane and with turbines on: no hardware fault occurred. 

However today we'll perform also take off tests and I would wait for these tests to be successful before closing the issue!

Uboldi80, 2022-07-27:
Well, it means that following the replacement of the EMS4 EB9 the problem has been solved.

closed 2022-07-27 by Uboldi80