robotology/mesh-support #1391: Right front FTs IMU do not broacast any data

GiulioRomualdi, 2022-07-04:
### Device name 🤖

iCubGenova09

### Request/Failure description

we tried to log the IMU data of the strain2 IMU mounted on the iCubGenova09 feet. We noticed that all the sensors stream accelerometer, gyro, and Euler data except the Right front FTs

### Detailed context

We modified the robots-configuration file as in the following commit https://github.com/ami-iit/robots-configuration/pull/17/commits/6cc02d73e7158589c3ff3731d8f456ee34160848

Once we run the robot interface we got the following warning.

```
<WARNING>  from BOARD 10.0.1.12 (right_leg-eb12-j3_5), src LOCAL, adr 0, time 45s 819m 961u: (code 0x00000039, par16 0x0005 par64 0x0000000000002000) -> SYS: a service has detected that some CAN boards are not broacasting anymore. In par16 the type of service, in par64 LS 32 bits the bit mask of lost board (CAN1 in MS 16 bits and CAN2 in LS 16 bits) + .
```

We tried to collect the data of the Right front FTs IMU with the `multianalogsensor` client/server infrastructure and we noticed that the data (ACC,gyro and eul) are constantly equal to zero. 

[image]

### Additional context

This is the log file

[log_icub-head_yarprobotinterface_1916_can2.txt](https://github.com/robotology/icub-tech-support/files/9038799/log_icub-head_yarprobotinterface_1916_can2.txt)

cc @S-Dafarra @pattacini @traversaro

pattacini, 2022-07-04:
Hi @GiulioRomualdi 
Just a clarification.

What's the flawed FT sensor? The rear or the front?

GiulioRomualdi, 2022-07-04:
> Hi @GiulioRomualdi Just a clarification.
> 
> What's the flawed FT sensor? The rear or the front?

Ops. Sorry @pattacini. It's the rear

pattacini, 2022-07-04:
Is the FT data being streamed out correctly from the right-rear sensor?

GiulioRomualdi, 2022-07-04:
To give a complete picture. the strain2 firmware is `v2.14`

Uboldi80, 2022-07-05:
I replaced the ft sensor located in the right foot on the front.
now the IMU data stream is working correctly.

pattacini, 2022-07-05:
Hi @Uboldi80 
The flawed sensor was the rear-right, but you said above that you replaced the front-right.

Just reopening the issue for clarification.

Uboldi80, 2022-07-05:
Thanks @pattacini, I corrected the misunderstanding.

I replaced the rear sensor.

closed 2022-07-05 by Uboldi80

GiulioRomualdi, 2022-07-05:
Thank you @Uboldi80 for taking care of the issue