robotology/mesh-support #1394: Left arm oscillates in position control thus having an asymmetric behaviours with respect to right: loose tendons?

DanielePucci, 2022-07-08:
### Device name 🤖

iCubGenova04 (iRonCub)

### Request/Failure description

Yesterday, I had the chance to perform some experiments with the iRonCub team. One experiment we did was to put the robot in position control, arms straight, and then push a little both arms in a symmetric way. The overall effect is that the **left** arm  starts oscillating, the right no. We checked, the control gains are the same. 

Probably, to check, it may be that tendons are not tight. So, this could be an easy one.

CC
@gabrielenava 

### Detailed context

See above.

Uboldi80, 2022-07-12:
yesterday I replaced the 2 FT present on the arms taking them from strain 1 to strain 2, also I solved the play problem on the left arm due to the screws on the flex wave that were loose.
this morning with the support of @Gandoo the can cable that from 2 foc 0B9 goes to the EMS EB9 has been replaced and the 10E1 cable has also been shortened and taken up again.

gsisinna, 2022-07-12:
The new strain2 sensors are configured and ready to be used. I tried to stream and read some data with @vpunithreddy for the left and right arms.

I created files like [this](https://github.com/robotology/robots-configuration/blob/master/iRonCub-Mk1/wrappers/FT/left_foot-FT_wrapper_multipleSens.xml) one for including a `multipleanalogsensorsserver`.

cc @DanielePucci

DanielePucci, 2022-07-12:
Thanks a lot @gsisinna 

CC
@HosameldinMohamed

gabrielenava, 2022-07-14:
the problem on left arm seems to be fixed, however we did not test for long time because a tendon broke, see #1399 

I think we can close the issue, other problems related to tendons will be tracked in #1399

closed 2022-07-14 by Uboldi80