robotology/mesh-support #1396: iCubGenova01 - Left thumb opposition behaves incorrectly

xEnVrE, 2022-07-09:
### Device name 🤖

iCubGenova01

### Request/Failure description

During some experiments, executed on Wednesday July 6th, the fingers started to move erratically and the left thumb opposition went beyond the hardware limits. We faulted the robot. The next time we tried to start the robot the problem presented again at calibration time.

In order to made the robot available, without the thumb, I tried to remove all the thumb joints, i.e. `0`, `1` and `2`, from the calibration. However, this way the whole forearm + hand did not complete the calibration. For this reason, as the robot was required for other tasks where the arms are not involved, at the moment the entire left arm has been disabled from the `icub_all_no_legs.xml` file.

It seems to me that this is the very same problem we experienced and described in https://github.com/robotology/icub-tech-support/issues/1335 (where you can also find a video).

### Detailed context

During some experiments, executed on Wednesday July 6th, the fingers started to move erratically and the left thumb opposition went beyond the hardware limits. We faulted the robot. The next time we tried to start the robot the problem presented again at calibration time.

In order to made the robot available, without the thumb, I tried to remove all the thumb joints, i.e. `0`, `1` and `2`, from the calibration. However, this way the whole forearm + hand did not complete the calibration. For this reason, as the robot was required for other tasks where the arms are not involved, at the moment the entire left arm has been disabled from the `icub_all_no_legs.xml` file.

It seems to me that this is the very same problem we experienced and described in https://github.com/robotology/icub-tech-support/issues/1335 (where you can also find a video).

Uboldi80, 2022-07-15:
This morning, following some checks, I found some problems related to corn.
The need is to review and replace some electrical wiring of the fingers that tend to restart the Corn during operation or do not allow the correct start of the setup.
given the complexity of the intervention and the time needed to resolve the issue, we will intervene after the Program increment, agreeing with the users to identify the most appropriate time.

xEnVrE, 2022-08-01:
Dear @Uboldi80, would this week be a candidate one for checking the robot?

Thank you

gsisinna, 2022-08-03:
This morning I performed several left hand tests to better investigate the problem mentioned by @xEnVrE. The left thumb behaves strangely when joints 5, 6 or 7 are forced or moved to strange positions. I checked all electrical connections and there are some problems with the hall sensor readings for joint 8. The same situation of wrong readings for joint 8 occurs when we hold joints 5 and 6 in place, and force the wrist to move by hand, thus increasing the current draw.

gsisinna, 2022-09-08:
Today I again checked all connections and replaced the MC4 responsible for the thumb oppose movement. The left arm starts, but strange behaviors regarding joint drift are still present, it remains to be investigated if the presence of the new fingertips can cause such problems.

cc @xEnVrE

gsisinna, 2022-09-12:
Further tests were conducted today. Joints 7,8,9 and 10 (left hand) were isolated from turn to turn, and it seems that the only sub-optimal way to have the left arm working is to eliminate joint 7 (hand finger) from the calibration. In this way the behavior of the thumb oppose seems more stable. The palm electronics have been revised again, and there are no short-circuit or other problems. I have also tested several yarprobotinterface by electronically removing the new fingertips, but they do not seem to play a role in the behavior.

I'll leave it to @xEnVrE and @fedeceola further testing for feedback on this!

xEnVrE, 2022-09-12:
Thanks for the update @gsisinna

xEnVrE, 2022-10-05:
After several experiments, we noticed that problem is still there, i.e., there are cases in which the encoder readings of the thumb makes it move beyond its hardware limits. This happens also when the joint 7 is disabled.

Fabrizio69, 2022-12-14:
iCubGenova01 will be dismissed soon, hence closing.

closed 2022-12-14 by Fabrizio69