robotology/mesh-support #1402: General maintenance

ddetommaso, 2022-07-21:
### Device name 🤖

iCubErzelli01

### Request/Failure description

We would need a general check on robot arms, torso and head possibly before the end of August.

### Detailed context

- Overall tightening of cables, electrical connections and other generic check up.
- Check range of motion of JOINT 4 for right and left arm.
- Reduce vibration on both robot arms.
- Face expressions module often doesn't work. Need a few restarting of the robot to make it work.

gsisinna, 2022-07-22:
With @ddetommaso we planned the check after the 8th of August.

cc @Nicolasr3

gsisinna, 2022-08-09:
Today I performed a general check of the robot, and the above forearm was calibrated to eliminate the asymmetry with the other arm. Several cables from the wrist and left shoulder were tightened, but there do not seem to be any obvious problems affecting the functioning of the robot.

Thanks @ddetommaso @Nicolasr3

closed 2022-08-09 by gsisinna