robotology/mesh-support #1404: ``r_wrist_yaw`` encoder diverges

S-Dafarra, 2022-07-22:
### Device name 🤖

iCubGenova09

### Request/Failure description

The ``right_wrist_yaw`` encoder seems to diverge. When we started the robot, the joint was ok, but after about an hour, it reads 2196 degrees

### Detailed context

[image]

cc @mebbaid @GiulioRomualdi @isorrentino

mebbaid, 2022-07-22:
Perhaps this is linked to #1401

gsisinna, 2022-07-25:
Today we checked all the magnets in the right wrist joints and replaced one. We re-calibrated joints 5 and 6 by removing the old deltas and everything seems fine!

cc @S-Dafarra

closed 2022-07-25 by gsisinna