robotology/mesh-support #1405: Mechnical issue on the iCubGenova09 neck

S-Dafarra, 2022-07-27:
### Device name 🤖

iCubGenova09

### Request/Failure description

The control of the iCubGenova09 neck sometimes has large errors, especially when applying a force to the head or when walking. It seems like some pully is slipping or the cables are loose.

### Detailed context

https://user-images.githubusercontent.com/18591940/181305555-5a8fcf65-f4b7-427d-b892-6d89a0f50211.mp4

gsisinna, 2022-07-28:
Today we analyzed the behavior shown in the video, and it appears to be a problem of relative sliding of the steel cable on the mechanism for the degree of freedom of the neck pitch.

As soon as @isorrentino has finished his experiments we will turn to the problem.

cc @S-Dafarra

gsisinna, 2022-08-01:
@S-Dafarra did you have a chance to check the functioning of the neck after my colleagues @fbiggi and @Uboldi80 improved the tensioning? If that doesn't work out, the whole mechanism will have to be opened.

S-Dafarra, 2022-08-01:
I don't know if something else has been done since I left, but I remember the problem was still there on Thursday. I am not going to have a chance to test the robot any time soon, but you could try yourself by applying a force on the head with the robot powered up.

isorrentino, 2022-08-01:
>@S-Dafarra did you have a chance to check the functioning of the neck after my colleagues @fbiggi and @Uboldi80 improved the tensioning? I

Today @mebbaid  and @paolo-viceconte tested the walking for 10 minutes to test the new battery. Maybe, they can answer your question.

mebbaid, 2022-08-02:
We did not pay close attention to the neck joints behavior. Perhaps @paolo-viceconte have noticed something ?

paolo-viceconte, 2022-08-02:
> We did not pay close attention to the neck joints behavior. Perhaps @paolo-viceconte have noticed something ?

Not really, sorry. We just focused on the walking-with-battery tests.

S-Dafarra, 2022-08-17:
Hi @gsisinna, I just checked the robot neck and unfortunately the problem is still there.

GiacomoVassallo13, 2022-08-31:
Buongiorno le wave generator dell' Harmonic drive sono stati smontati e ripuliti poiche il grasso era pieno di sporcizia rimasta incollata. Questi sono stati puliti con la solita procedura, ossia con un bagno d'alcool poi sono stati riunti onde evitare ossidazioni e ingassati. Svolta questa procedura le wave generator sono state rimontate. Ora il colo presenta movimenti decismente piu fluidi. Attendo feedback per sapere se il problema e' effettivamente risolto.

cc: @S-Dafarra @DanielePucci

S-Dafarra, 2022-08-31:
> Attendo feedback per sapere se il problema e' effettivamente risolto.

Si il problema sembra essere risolto!

closed 2022-09-01 by gsisinna