robotology/mesh-support #1407: Trip damages. Broken thumb distal tendons

pabloreyesrobles, 2022-08-02:
### Device name 🤖

iCubValparaiso01

### Request/Failure description

Harsh trip apparently:
- Left eye cam lens fell when taking robot out of the box.
- Right eyebrow detached from its position.
- Eyelashes movement blocked by face piece. Need re-calibration.

Both distal thumbs tendons are damaged, details in next section.

### Detailed context

We started using the iCubValparaiso01 recently since its arrival. As soon as we removed the robot from the box, the lens of the left eye fell out, without apparent damage. It was installed back in position and the image is showing fine but requires recalibration.

Once turned on we noticed that the right eyebrow was not in its position and the eyelashes had difficulty moving, apparently being pressed by the face piece. It was removed to rearrange the eyebrow. We have encoder reading from the blinking system but it requires recalibration.

Finally in motion testing (yarpmotorinterface only) we noticed some initialization issues with some motor controllers on the left arm, apparently due to joints like the elbow being out of range. A screenshot of the yarpmotorgui during the event is displayed in the shared folder. We flexed the elbow and went into idle state, running yarpmotorinterface again. The left hand was closed with a delay except for the left thumb, which after verification we noticed that the distal tendon was cut.

In a later test we ran the yarpmotorinterface again and heard a cut on the right arm this time, this time the distal thumb tendon of that hand. This situation is particular because we have encoder readings of said joint.

### Additional context

Log of the startup and left arm being rearranged passing to idle state.
[yarprunlog_25_07_2022_15_15_50.log](https://github.com/robotology/icub-tech-support/files/9244205/yarprunlog_25_07_2022_15_15_50.log)

Folder with images and screenshots of some of the issues reported.
https://usmcl-my.sharepoint.com/:f:/g/personal/pablo_reyesr_alumnos_usm_cl/Epj9hBL1ERVDrsHXg3BdJeoBKl9AHkkCxhPpgQQXnilXFw?e=mQsGzR

If more info/images/logs needed please let me know.
cc @mjescobar @Rlanasg

gsisinna, 2022-08-03:
Hi @pabloreyesrobles, we are very sorry to hear that you had problems with the robot after transport. Going in order, I will try to provide you with everything you need to be able to fix the robot and get it back to a state of optimal functionality.

**Thumb distal**
For thumb distals you can find the complete replacement guide at the following [link](https://icub-tech-iit.github.io/documentation/icub_tendons/lower_arm_V2/#thumb). We saw this procedure in the laboratory when you came to our center if you remember.

**Eyelashes**
For the eyelashes I imagine that the robot suffered some frontal damage during transport and the camera shifted, preventing proper movement over the lenses. Here a picture:
[image]
The depth of the camera protruding from the eye can be changed by rotating the lens by hand, and this will also change the focus, which must be calibrated as in the other eye. So I would perform this procedure with at least the cameras active. 

If you should not be able to rotate the chamber then you have to disassemble the white circular cover (by rotating it) of the eye and have access to the holder where the chamber is attached with screws.

[image]

**Eyebrown**
Finally for the right eyebrown this should be attached directly to the face of iCub (white mask), so we need to see where the PCB support was damaged possibly. Could you disassemble the face cover and send a picture of the situation? I would remind you to always handle components of this type with care and not to pull electrical cables from the head.

**Calibration**
For all types of joint calibration I refer you to the full reference: 
[Robot Calibration of iCub v2.X](https://icub-tech-iit.github.io/documentation/icub_robot_calibration/icub-robot-calibration-v2.x/?h=calibration#robot-calibration-of-icub-v2x)

We remain available for anything, thank you!

cc @maggia80 @Fabrizio69

pabloreyesrobles, 2022-08-03:
Hi @gsisinna, thank you for the quick response.

**Thumb distal**
What is the procedure for requesting the related cables?

**Eyelashes**
It seems that the issue is related to the depth of the cameras as you are predicting. Here is a picture:
[image]

**Eyebrows**
Here are some pictures of the face cover. Looks like the PCB board doesn't suffer any damage, plus we were able to run the face expressions examples.
[image]
[image]

One more thing I forgot to mention is that there is some parts with some rust on it. Any recommendations to take care of it?
[image]
[image]

Thank you in advance!

gsisinna, 2022-08-04:
For cables you can email @maggia80 and [Andrea Derito](https://www.iit.it/it/web/guest/people-details/-/people/andrea-derito) directly for the parts listed [here](https://icub-tech-iit.github.io/documentation/icub_tendons/lower_arm_V2/#22-wiring) with code:

**IITCODE**: 2391
**Alias**: U7194561
**Description**: Stainless steel micro-cable, cable construction 7x19mm, nom. diameter 0.45mm, outer diameter 0.61mm, Polyamide coated, AISI 316 - CARL STAHL GMBH

Regarding face expressions, if the PCB support is intact, I would try reattaching the boards to the face cover.

While for the **rust**, it is very infrequent what you sent me by photo, and it makes me think that the robot has suffered of humidity during transportation. This problem should be better investigated, and I will consult with my colleagues to suggest the most appropriate treatment.

pabloreyesrobles, 2022-08-08:
I was rotating the camera lenses to increase their depth. As you commented the focus changed and now I get a blurry image on both cameras. Do I have to follow the procedure in the [calibration instructions](https://icub-tech-iit.github.io/documentation/icub_robot_calibration/icub-robot-calibration-v2.x/?h=camera#cameras-calibration) to refocus the images?

gsisinna, 2022-08-09:
@pabloreyesrobles The focus is adjusted manually through the rotation of the lens. While on the software side it is possible to act on fish-eye elimination. I would try to find a compromise between lens depth and proper focus, so that the whole mechanism can close.

gsisinna, 2022-09-07:
Hi @pabloreyesrobles, any news?

gsisinna, 2022-09-12:
Hi @pabloreyesrobles, 
I have consulted my colleagues at the workshop and the best solution would be to disassemble the part in question and do a complete cleaning with solvents specifically for that purpose. This is not a simple operation, so I am wondering if this rust impacts only aesthetically or also functionally for you.

I will also communicate your request for spare parts.

If you want, we can also arrange a videocall these days to see the remaining problems together. Thanks!

pabloreyesrobles, 2022-09-12:
Hi @gsisinna,
It seems that the rust doesn't impact functionally, at least for now, but I think that eventually that rust should be cleaned/removed so it doesn't spread.

The videocall sounds great. I'm available wednesday and thursday. Should morning here in Valparaiso and afternoon there in Genova. Let me know what time suits you.

gsisinna, 2022-09-22:
Hi @pabloreyesrobles, sorry for the late reply. I will be available tomorrow (September 23) or from next week on the days you mentioned.

Thanks!

pabloreyesrobles, 2022-09-22:
Hi @gsisinna. I think September 29 at 10:30 (15:30 IT) is ideal for us. Let me know if it suits your schedule.

gsisinna, 2022-09-23:
yes @pabloreyesrobles, I sent an invitation!

gsisinna, 2022-09-30:
Hello @pabloreyesrobles, as already mentioned during the videocall I am waiting to hear from you regarding the receipt of spare parts for thumb repairs and I remain available to guide you through the procedure of reinserting thumbs into calibrator files.

pabloreyesrobles, 2022-10-03:
Hi @gsisinna, we already received the parts. We could schedule the meeting for this thursday at the same time as the previous one, if that suits you. Let me know!

gsisinna, 2022-10-07:
Closing after the remote support done on thursday 06/10/22, thanks!

closed 2022-10-07 by gsisinna