robotology/mesh-support #1408: [iCubGenova09] Torso pitch does not move even if in position control

isorrentino, 2022-08-03:
### Device name 🤖

iCubGenova09

### Request/Failure description

While testing the walking the robot fell down. The torso went in HF. We tried to switch back in position control also rebooting the motors. The torso seems to be controlled but the joint does not move. 

The encoder values changes if the link is manually moved

### Detailed context

[log_icub-head_yarprobotinterface_3652.txt](https://github.com/robotology/icub-tech-support/files/9250606/log_icub-head_yarprobotinterface_3652.txt)

cc @GiulioRomualdi

gsisinna, 2022-08-04:
I found the 2FOC of the torso completely detached and with some damaged connectors, I think during the fall and without the battery supporting the power connector something was pulled.

We will check all the wiring together with @emilianob80  to see where the problem is.

emilianob80, 2022-08-08:
I encountered two problems on the WTP1 cable.

1) red and black wires, sensor side, almost completely broken inside the connector

2) black thread crushed and then damaged

I have completely replaced the cable.

@gsisinna If everything works, you can close the issue

[image]
[image]

cc: @isorrentino @Fabrizio69

mebbaid, 2022-08-08:
Unfortunately, the issue seems to persist. When starting the `yarprobotinterface` the `torso_pitch` does not calibrate but starts in `idle`. Even when we switch to position control, we are unable to move the torso pitch. Here is the output of the logger and also a picture of the calibration from the `yarpmotorgui`

[log_icub-head_yarprobotinterface_2691_August_08.txt](https://github.com/robotology/icub-tech-support/files/9281039/log_icub-head_yarprobotinterface_2691_August_08.txt)

[image]

emilianob80, 2022-08-09:
@mebbaid After a further check of the wiring, I considered it appropriate to totally replace the WE1 cable because it is too short and impossible to re-crimp in case of breakage. On the card side, the white wire broke during the check. It is difficult to understand the cause (it was already broken or broke during the manipulation of the connector) because it was drowned in the glue.

Hope the problem has been solved

cc: @gsisinna @Fabrizio69  @Uboldi80

mebbaid, 2022-08-09:
thanks @emilianob80  the `yarprobotinterface` ran smoothly this time, and we were able to move the `torso_pitch`.

emilianob80, 2022-08-09:
@gsisinna chiudo la issue

 @Fabrizio69 @Uboldi80

closed 2022-08-09 by emilianob80