robotology/mesh-support #1409: Right shoulder tendons exited

HosameldinMohamed, 2022-08-05:
### Device name 🤖

iCubGenova04

### Request/Failure description

While the robot was moving the arms for FTs in-situ calibration test, the tendons exited from the right shoulder.

### Detailed context

Please see the picture below.

### Additional context

[image]

### How does it affect you?

It blocks Arms FTs in-situ calibration experiments.

maggia80, 2022-08-05:
@HosameldinMohamed if no one is available you could check the documentation here and try to follow it: https://icub-tech-iit.github.io/documentation/icub_tendons/shoulder_elbow/

HosameldinMohamed, 2022-08-05:
> @HosameldinMohamed if no one is available you could check the documentation here and try to follow it: https://icub-tech-iit.github.io/documentation/icub_tendons/shoulder_elbow/
> 
> 

Great, thanks!

HosameldinMohamed, 2022-08-05:
Cable put back with @vpunithreddy
[image]
!

closed 2022-08-05 by HosameldinMohamed

HosameldinMohamed, 2022-08-09:
Apparently, our intervention caused some problems. We did not put the motors in the initial position when putting back the tendons.

We will write more details about it.

The problem was found and fixed by @GiacomoVassallo13, see https://github.com/robotology/icub-tech-support/issues/1412#issuecomment-1209419807