robotology/mesh-support #1410: Left arm tendon broken

ddetommaso, 2022-08-05:
### Device name 🤖

iCubErzelli02

### Request/Failure description

One of the tendon of the wrist (left arm) broke during a demo.

### Detailed context

(please if needed ask us for further investigations)

### Additional context

[image]

gsisinna, 2022-08-08:
Today I disassembled the arm in question as it is necessary to recreate custom cables for the active wrist differential mechanism.

cc @Uboldi80 @Fabrizio69 

[image]

gsisinna, 2022-08-09:
Yesterday with @GiacomoVassallo13's help we regenerated the wrist cables by creating new custom ones and restored the differential mechanism:
[image]
Here is a video of the calibration and proper functioning of the left wrist:

https://user-images.githubusercontent.com/36999962/183647038-629b72fd-df8b-464d-8f35-1557c147d08c.mp4

cc @ddetommaso @Nicolasr3 @Fabrizio69 @maggia80

closed 2022-08-09 by gsisinna