robotology/mesh-support #1411: Torso often goes in hardware fault

HosameldinMohamed, 2022-08-07:
### Device name 🤖

iCubGenova04

### Request/Failure description

The robot either starts with the torso already in hardware fault or after a short while it goes into hardware fault.

### Detailed context

The days before, the robot was ok,
I started the robot while being at the configuration shown the picture below, which I usually do when it's on the pole. The robot had a spark around the torso area and the and the boards/motors button started to flash red light.

I restarted the configuration and kept the robot at a vertical configuration while starting, and it started without issues. After a long time, when I started to do some motions with the legs, the torso went into hardware fault with the following messages

```
	8834,278012	ERROR			 from BOARD 10.0.1.5 (torso-eb5-j0_2), src CAN1, adr 0, time 328s 153m 121u: (code 0x00000019, par16 0x4001 par64 0x0000000000000000) -> SYS: EOtheCANservice could not tx frames on CAN bus. In par16 there is: on msb the size of txfifo, on lsb a code. + .

	8834,278001	ERROR			 from BOARD 10.0.1.5 (torso-eb5-j0_2), src LOCAL, adr 0, time 328s 197m 427u: (code 0x00000013, par16 0x0005 par64 0x0000000000000000) -> SYS: the EOtheInfoDispatcher could not accept a eOmn_info_properties_t item inside its transmitting queue. In par16 there is the number of lost items. + .
```

The fault wouldn't go away until I restarted the robot, then the same problem started to happen more frequently and when starting either it starts with the fault or starts fine but the fault happens after some legs movements.

### Additional context

Robot state when first startup:
[image]

Here's the complete log file:

[yarprunlog_07_08_2022_19_57_11.log](https://github.com/robotology/icub-tech-support/files/9277282/yarprunlog_07_08_2022_19_57_11.log)

HosameldinMohamed, 2022-08-08:
CC @gsisinna

HosameldinMohamed, 2022-08-08:
With @vpunithreddy we were able to trigger the issue by moving the **right leg pitch joint** back and forth. We moved it to the back until the end and then moving forward while reaching around `0deg`, the error was triggered.

We were looking at the wires in the back but we did not find anything strange (like an exposed cable).

gsisinna, 2022-08-10:
Today I tried to check the boards for the torso joints and I found a screw inside the 2FOC holder. After removing the screw there are no more sparks and the torso does not go into Hardware Fault.
I am waiting for some tests from the guys to see if this is the cause of the problems.

cc @HosameldinMohamed @vpunithreddy

HosameldinMohamed, 2022-08-16:
Thanks @gsisinna!

> I am waiting for some tests from the guys to see if this is the cause of the problems.

We did further tests on the robot and the spark and hardware fault issues did not happen again.
We started the robot and moved the joints around using `yarpmotorgui`, we also ran the FT-calibration script that triggered this fault before. All is OK!

The only issue currently is when starting the robot being on the pole, inclined to the back. The motors don't start, as shown in the following video:

https://user-images.githubusercontent.com/45564317/184877680-b356aa63-9b1f-43ae-8a05-89b0206e9555.mp4

Holding the robot in a straight configuration doesn't trigger this issue, and the robot starts with no problems.

not sure but I see the back harness being pressed when the robot is inclined on its back.

CC @gabrielenava

HosameldinMohamed, 2022-09-01:
> not sure but I see the back harness being pressed when the robot is inclined on its back.

Checking the harness further, it seems that there's some damage to the + and - of a power cable inside. It appears when bending the robot forward, see the picture below:

[image]

If you look closer, you can see some of the copper wire is exposed from both the red and black cables 😮. They are very close but are not touching, but when the robot is inclined to the back, the metal part (of the torso) touches both cables forming a short circuit. That's why the problem appears only in that configuration!

[image]

HosameldinMohamed, 2022-09-02:
CC @vpunithreddy @gabrielenava

gsisinna, 2022-09-07:
Cables rebuilt from new, it remains to test the robot with the team guys to make sure the power supply is working properly

cc @gabrielenava @vpunithreddy

gsisinna, 2022-09-07:
Robot tested and working. An Ethernet line on EB3 also had to be fixed as it was broken.

closed 2022-09-07 by gsisinna