robotology/mesh-support #1412: Shoulder Yaw cable broken - right arm

HosameldinMohamed, 2022-08-09:
### Device name 🤖

iCubGenova04

### Request/Failure description

The robot was faulted and the right arm fell down with the jet. Then I saw the Yaw cable broken.

### Detailed context

According to [the documentation](https://icub-tech-iit.github.io/documentation/icub_tendons/shoulder_elbow/#21-left-shoulder-rotation-cable-1-and-2). The broken cable is **number 8**. See the picture below.

### Additional context

[image]

### How does it affect you?

It blocks FTs in-situ arms calibration experiments.

HosameldinMohamed, 2022-08-09:
Cc @GiacomoVassallo13

GiacomoVassallo13, 2022-08-09:
Hi, the cable 8 of right shoulder was been replaced. But during the operation I noticed the cable 1 of the same shoulder which was not wired correctly, due to an unsuccessful intervention by the guys who tried to wire it without support.
@HosameldinMohamed  point out the points in the guide that are not clear so as to correct them.
currently everything has been restored and the cables tensioned. I continue with the closure of the issue.

cc: @HosameldinMohamed @Fabrizio69

HosameldinMohamed, 2022-08-09:
CC @vpunithreddy

HosameldinMohamed, 2022-08-09:
Thanks @GiacomoVassallo13