robotology/mesh-support #1416: Right index distal broken

S-Dafarra, 2022-09-01:
### Device name 🤖

iCubGenova09

### Request/Failure description

The right index distal tendon got broken

### Detailed context

We were doing some endurance tests and the tendon broke

Fabrizio69, 2022-09-06:
@GiacomoVassallo13 in the motor replacement (new motor is cod.2471) remember if doable, a tendon with a bigger diameter (but w/o teflon, according to [issue](https://github.com/icub-tech-iit/task-force-miscellanea/issues/132#issuecomment-1238165774)). Thanks

cc: @salvi-mattia, @gsisinna

salvi-mattia, 2022-09-07:
ciao @Fabrizio69 ,
il gruppo che mi ha consegnato 
[image]

corrisponde al codice 3509 . A2 -SINGLE TENDON FINGER MOTOR ASSEMBLY  - RC_TLR_010_G_009

su wingst dal quale dovresti essere in grado di dedurre l'intera lista parti
(ovviamente il motore e' diverso)

fammi sapere se queste info ti sono sufficienti

salvi-mattia, 2022-09-07:
@Fabrizio69 
here the MWS request for the two capstain we used

[image]

GiacomoVassallo13, 2022-09-08:
Buongiorno, l' intrvento per ripristinare il distale montando un motore piu potente e' stato eseguito. Il tendine e' stato relaizzato con cavo cod. 1815 di diametro superiore rispetto al cavo montato di solito sui distali ( cod. 2381), affinche possa reggere al maggiore carico generato dal motore piu potente. Il cavo in questione a differenza del cod. 2381 non e' ricoperto in teflon, cio' puo portare a maggiori atriti comportando un movimento meno fluido del polso e andando a consumare le pulegge delle dita. Appena mi sara' possibile allego il video del test di movimento.
Allego ticket di materiale : https://github.com/icub-tech-iit/tickets/issues/2459

cc: @salvi-mattia @Fabrizio69 @maggia80

Fabrizio69, 2022-09-09:
>
> Appena mi sara' possibile allego il video del test di movimento.
> 

Ciao @GiacomoVassallo13 il video?

gsisinna, 2022-09-12:
https://user-images.githubusercontent.com/36999962/189689369-3a6ee770-5d60-4dae-85be-a129f81e32ae.mp4

Grazie @GiacomoVassallo13.

cc @Fabrizio69