robotology/mesh-support #1417: Left wrist yaw encoder diverges

S-Dafarra, 2022-09-01:
### Device name 🤖

iCubGenova09

### Request/Failure description

After starting the robot, the joint goes almost immediately in HF because the encoder diverges.

### Detailed context

The encoder value diverges quickly to several thousand degrees.

S-Dafarra, 2022-09-05:
Edited the title of the issue as this happened also on the right. See the following graph. I was not touching the arm at all (i was pushing on the legs)

[image]
[image]

gsisinna, 2022-09-06:
Hi @S-Dafarra, have you seen this behavior after the check performed on the AEA boards?

S-Dafarra, 2022-09-06:
> Hi @S-Dafarra, have you seen this behavior after the check performed on the AEA boards?

Hi @gsisinna, I haven't noticed since then!

closed 2022-09-07 by gsisinna