robotology/mesh-support #1422: Left arm shoulder pitch and elbow tendons broke

HosameldinMohamed, 2022-09-08:
### Device name 🤖

iCubGenova04

### Request/Failure description

During F/T sensor calibration activities with additional weights, the arms tendons broke down again. The broke joints are the shoulder pitch and the elbow. I think the shoulder roll tendons exited too.

### Detailed context

We are stressing the robot with this experiment, even though the weight that broke the arm was tested before, we were repeating the tests increasing the weights gradually to reach that specific value. That probably wore the tendons out.

### Additional context

This piece fell off the robot

[image]

HosameldinMohamed, 2022-09-08:
CC @gabrielenava @vpunithreddy

gsisinna, 2022-09-12:
We replaced the two cables with @GiacomoVassallo13 and the robot has been tested with @HosameldinMohamed.

closed 2022-09-12 by gsisinna

HosameldinMohamed, 2022-09-12:
Thanks guys!!