robotology/mesh-support #1423: right wrist hardware fault

Nicolasr3, 2022-09-12:
### Device name 🤖

iCubErzelli02

### Request/Failure description

After running the module yarprobotinterface, during the calibration the right wirst never reached his position. 

[image]

### Detailed context

We are perfoming an experiment where iCub needs to press several buttons in front of it, with the right hand.

### Additional context

Here some picture of the state of the cables wirst.
[image]
[image]
[image]
[image]
[image]

The yarpmotorgui is plotting this error on the joint 6 of the right arm:

[image]

### How does it affect you?

Now we stopped the data collection of the experiment until the robot wirst is repared.
It would be really helpfull if the issue could be solved soon, so we can restart the data collection.

gsisinna, 2022-09-12:
Hi @Nicolasr3,
I will be there tomorrow 👍

The robot can be left in the room where you currently use it and connected to the network.

gsisinna, 2022-09-13:
This morning I replaced the AEA electronic board for the wrist yaw movement. Now the robot is working again and below is the video of the calibration including the wrist:

https://user-images.githubusercontent.com/36999962/189867902-ab3bad9f-66d6-424a-84ea-93368d0a6ff4.mp4

cc @Nicolasr3 @ddetommaso @Fabrizio69 @maggia80

closed 2022-09-13 by gsisinna