robotology/mesh-support #1424: [iCubGenova09] Unable to reach the icub-head

isorrentino, 2022-09-13:
### Device name 🤖

iCubGenova09

### Request/Failure description

I switched on the icub-head but I'm not able to reach it both in wifi and ethernet configuration. 

### Detailed context

I tried to ping the head and to access it in SSH but nothing works.

cc @pattacini @DanielePucci @S-Dafarra

GiulioRomualdi, 2022-09-13:
We also tried to reboot the server. The problem is still there

GiulioRomualdi, 2022-09-13:
@pattacini @maggia80 and @Fabrizio69 came to check. After connecting the head to the monitor they noticed that Ubuntu was rebooting while starting. @pattacini @maggia80 and @Fabrizio69 noticed that the maximum power was set at 30W. They increased it up to 1500W and they blocked the power supply UI. 

Thank you all

closed 2022-09-13 by GiulioRomualdi