robotology/mesh-support #1426: R1SN003 neck pitch does not work properly

elandini84, 2022-09-15:
### Device name 🤖

R1SN003

### Request/Failure description

R1SN003 neck cannot go upwards and when going in the opposite direction is it moves past the wanted amount of degrees.

### Detailed context

We tried different times to move R! head downwards and then up again but it never worked. The head just stays down unless some moves it manually in that direction.

Fixing this issue is pretty important since on the 20th of September (next Tuesday) we have an important demo.

fbiggi, 2022-09-15:
ciao @elandini84 
domani mattina intervengo su r1

cc @Fabrizio69 @maggia80

fbiggi, 2022-09-16:
repositioning and adjustment of the neck strap
testing performed together with @ elandini84
total hours 3:25 minutes

cc @Fabrizio69 @maggia80

elandini84, 2022-09-16:
The neck now works as expected, thanks a lot @fbiggi

closed 2022-09-16 by elandini84