robotology/mesh-support #1427: Problems on the right thumb proximal, right index and left pinky

GiulioRomualdi, 2022-09-16:
### Device name 🤖

iCubGenova09

### Request/Failure description

While testing the teleoperation system, we noticed the following problems:
1. the **right thumb proximal** often goes in hardware fault 
2. The **left ring finger** keeps going at 90 deg even if the left_piny measures zero
3. It is not possible to completely close the **right index**

### Detailed context

Please check above

### Additional context

cc @S-Dafarra @isorrentino @RiccardoGrieco

S-Dafarra, 2022-09-19:
> * It is not possible to completely close the **right index**

The right index distal, as it happened last week.

paolo-viceconte, 2022-09-19:
Updates after today's intervention by @emilianob80:

> While testing the teleoperation system, we noticed the following problems:
> 
> 1. the **right thumb proximal** often goes in hardware fault

Due to glue-related issues, the encoder was not reading properly. @emilianob80 fixed it. The issue is not solved though since the cable of the thumb proximal (motor 3m1) is weirdly placed. This is a mechanical issue and needs to be addressed.

> 2. The **left ring finger** keeps going at 90 deg even if the left_piny measures zero

The left pinky and ring fingers touch each other during calibration and therefore the joint often ends the calibration in NOT CONFIGURED. By helping the calibration manually (i.e. keeping the two fingers more distant each other) the calibration succeeds but the ring finger does not reach the upper limit. In short, the left pinky calibration should be doublechecked.
 
> 3. It is not possible to completely close the **right index**

The index proximal often goes in HF (along with the thumb), so we could not even properly test this. Since the encoder does not read properly, a longer intervention needs to be scheduled.
 
> 4. The left hand skin does not work

The CAN cable of the e1B7seemed to be broken. @emilianob80 repaired it. Still the left hand skin is not working properly.

cc @S-Dafarra

emilianob80, 2022-09-20:
@paolo-viceconte This morning I proceeded to:
- replacement of the left little finger. Yesterday caused all fingertips not to work. I have tested the wires and there seem to be no further problems.
- replacement of hall sensor wires proximal index finger and relative sensor

It is necessary to carry out the function test

cc: @S-Dafarra @Fabrizio69 @gsisinna

emilianob80, 2022-09-20:
@paolo-viceconte @S-Dafarra in addition to the wires of the proximal hall sensor, the sensor was also replaced. The magnet was not touched and therefore I do not know if the calibration values have been changed

@Fabrizio69 @gsisinna

GiacomoVassallo13, 2022-09-20:
Hi the thumb proximal it was replaced. i'm waiting a feedback

cc: @S-Dafarra @gsisinna

gsisinna, 2022-09-21:
Today I restored the correct winding direction for the proximal thumb in accordance with the mechanical intervention made by @GiacomoVassallo13. Now this joint works correctly and has been tested several times together with @S-Dafarra.

New values for rotor position were also derived for the new DC motors of the index distal and middle distal, so as to avoid excessive cable windings and blockages in reaching then desired position. The values were derived by reading the raw motor position and dividing by the iCub degrees parameter (182.044), values approximated to the second decimal place.

The changes were committed and there is a record of them in the git logs.

Only for the pinky remains to redo the mechanical cable or set a new max and min position for the rotor, but this will be decided after the demos and analysis to be done with the robot today.

cc @Fabrizio69 @maggia80

S-Dafarra, 2022-09-21:
Here the commits:
- https://github.com/ami-iit/robots-configuration/commit/ead4aa92183533b1c3fca60ed085935bd8884d1b
- https://github.com/ami-iit/robots-configuration/commit/926d05cc22ae62dad229ebf1a7a0412d93d488b9
- https://github.com/ami-iit/robots-configuration/commit/3cdcd4f45681fcf3551fa700e55e15576619555c

S-Dafarra, 2022-09-22:
The remaining issues are:
- the left pinky that closes more than the middle
- the right middle finger skin does not work anymore (after some energetic handshake from some guests)
- some taxels of the left index finger remain active continuously

gsisinna, 2022-09-28:
Closing since the remaining problems have been solved:

> The remaining issues are:
> 
> * the left pinky that closes more than the middle
> * the right middle finger skin does not work anymore (after some energetic handshake from some guests)
> * some taxels of the left index finger remain active continuously

closed 2022-09-28 by gsisinna