robotology/mesh-support #1429: iCubGenova01 - Right hand palm skin and right forearm skin not working

xEnVrE, 2022-09-19:
### Device name 🤖

iCubGenova01

### Request/Failure description

As per the title, the skin of the palm of the right hand and that of the forearm do not respond to external stimuli

### Detailed context

As per the title, the skin of the palm of the right hand and that of the forearm do not respond to external stimuli

xEnVrE, 2022-10-05:
I updated the issue title as many other parts stopped working.

Specifically, at the moment, the following **do not work**:
- left arm: all parts
- right arm:
  - arm
  - forearm
  - palm
  - ring
  - medium (works but it has spikes every so often)

Fabrizio69, 2022-12-14:
iCubGenova01 will be dismissed soon, hence closing.

closed 2022-12-14 by Fabrizio69